ARMmbed/mbed-os #14510: PPP connection hangs with u-blox SARA-R500S

jdbruijn, 2021-04-07:
### Description of defect

Opening a PPP connection with the u-blox SARA-R500S hangs until it's timed out. I have noted two things:

1. APN logged as response on the `AT+CGDCONT?` command is `iot.t-mobile.nl.mnc040.mcc901.gprs`, while I've configured it as `iot.t-mobile.nl`.
  I believe this is no issue and simply some operator identifier, but please correct me if I'm wrong. I think I've read this somewhere but can't remember and quickly find where.
2. Weird characters after `CONNECT`.
  In the small log section below there are weird characters after the `CONNECT<cr><ln>` response to start PPP, which I don't see with the other modems we use (u-blox SARA-R410M and u-blox SARA-G350).

Please see the full log at the end, including the expected behaviour from the R410M log. We're using basically the same code with the u-blox SARA-R410M and are able to establish a PPP connection normally with the same SIM, provider and settings.

```
[INFO][CELL]: CellularContext PPP connect
[INFO][CELL]: AT TX ( 9): ATD*99***
[INFO][CELL]: AT TX ( 1): 1
[INFO][CELL]: AT TX ( 1): #
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 7): <cr><ln>CONNE
[INFO][CELL]: AT RX (27): CT<cr><ln>~<FF>}#<C0>!}!} } }6}"}&} } }
[DBG ][CELL]: ppp_status_cb: event 0, ptr 3
[DBG ][CELL]: callback: 0, ptr: 3
[DBG ][CELL]: process_oob, filehandle is not usable, return...
Connection timed out (180000 ms)
```

<details><summary>Same log section but with the u-blox SARA-R410M</summary>

```
[INFO][CELL]: CellularContext PPP connect
[INFO][CELL]: AT TX ( 9): ATD*99***
[INFO][CELL]: AT TX ( 1): 1
[INFO][CELL]: AT TX ( 1): #
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX (10): <cr><ln>CONNECT
[DBG ][CELL]: ppp_status_cb: event 0, ptr 3
[DBG ][CELL]: callback: 0, ptr: 3
[DBG ][CELL]: process_oob, filehandle is not usable, return...
[DBG ][CELL]: process_oob, filehandle is not usable, return...
[DBG ][CELL]: process_oob, filehandle is not usable, return...
[DBG ][CELL]: ppp_status_cb: event 0, ptr 1
[DBG ][CELL]: callback: 0, ptr: 1
Connection status: 3, cellular status: 24
Global IP received but waiting for local IP (48707 ms)
Connected as X.X.X.X in 48811 ms
```

</details>

Things I've already tried and additional observations.
1. PPP connect using `CGDATA`.
2. Using different MNO profiles (1, 90, 100).
3. We're connected/attached to the cellular network.
4. Additional logging shows that `nsapi_ppp_connect` is returning with OK code.

#### Target(s) affected by this defect ?

u-blox?

#### Toolchain(s) (name and version) displaying this defect ?

```shell
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

#### What version of Mbed-os are you using (tag or sha) ?

[... 597 lines cut ...]
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 2): <cr><ln>
[INFO][CELL]: AT RX (32): +CGDCONT: 1,"IP","iot.t-mobile.n
[INFO][CELL]: AT RX (32): l","X.X.X.X",0,0,0,0<cr><ln><cr><ln>OK<cr><ln>
[DBG ][CELL]: CID 1 APN "iot.t-mobile.nl" pdp_type 1
[INFO][CELL]: Found PDP context 1
[INFO][CELL]: CellularContext PPP connect
[INFO][CELL]: AT TX ( 9): ATD*99***
[INFO][CELL]: AT TX ( 1): 1
[INFO][CELL]: AT TX ( 1): #
[INFO][CELL]: AT TX ( 1): <cr>
[INFO][CELL]: AT RX ( 4): <cr><ln>CO
[INFO][CELL]: AT RX (17): NNECT 150000000<cr><ln>
[DBG ][CELL]: ppp_status_cb: event 0, ptr 3
[DBG ][CELL]: callback: 0, ptr: 3
[DBG ][CELL]: process_oob, filehandle is not usable, return...
[DBG ][CELL]: ppp_status_cb: event 0, ptr 1
[DBG ][CELL]: callback: 0, ptr: 1
Connection status: 3, cellular status: 38
Global IP received but waiting for local IP (9812 ms)
Connected as X.X.X.X in 9916 ms
[INFO][CELL]: CellularContext set blocking 1
```

</details>

jdbruijn, 2021-04-30:
Has anyone had a chance to have a look at this one (bump)?

adbridge, 2021-06-07:
@MarceloSalazar do you have any contacts at Ublox that might be willing to look at this ?

MarceloSalazar, 2021-06-07:
FYI @ARMmbed/team-ublox