ARMmbed/mbed-os #14528: STM32G474 - CAN TxIRq callback never called

pilotak, 2021-04-09:
### Description of defect

When setting up CAN with `TxIrq` callback (see example below), it's never called. `RxIrq` works as intended

#### Target(s) affected by this defect ?
STM32G474

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM 9.3.1

#### What version of Mbed-os are you using (tag or sha) ?

84e5f1e03eed2227d91937c38ab862297ee9c2d6

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 

```cpp
CAN can(PB_8, PB_9);
DigitalOut canStandby(PB_7, 0);
DigitalOut led(PB_14, 1);

void canTxDone() { // ISR
    led = !led;
}

int main() {
    can.frequency(250000);
    can.mode(CAN::Normal);
    can.attach(&canTxDone, CAN::TxIrq);

    while (1) {
        CANMessage msg;
        msg.id = 0;
        msg.id |= 6 << 26; // priority
        msg.id |= 0xFF00 << 8; // SPN
        msg.id |= 0xA0; // SA
        msg.len = 8;
        msg.type = CANData;
        msg.format = CANExtended;
        memset(msg.data, 0, 8);

        can.write(msg);
        ThisThread::sleep_for(1s);
    }
}
```

0xc0170, 2021-04-12:
@ARMmbed/team-st-mcd Please review

jeromecoutant, 2021-04-12:
@pilotak Is this issue related to https://github.com/ARMmbed/mbed-os/issues/9495 ?

pilotak, 2021-04-12:
@jeromecoutant No, mutex doen't play a role in this issue, as you can see the example above. The issue here is that the IRQ is not generated on `CAN::TxIrq`. 

**`CAN::RxIrq` is generated correctly** but as per issue #9495 it has a problem that you can't read messages in such a callback due to mutex (which can be overcome by disabling it) and you can't even use event queue for that due to too many IRQ requests if you are on fully loaded CAN bus ie.: in the car

pilotak, 2021-04-26:
Any chance of getting it fixed before complex issue #9495 is sorted out? i'm happy to provide more details if you guide me what you need

jeromecoutant, 2021-04-27:
Sorry I didn't check,
but sometimes IRQ issue comes from C/C++ mix in the code?

pilotak, 2021-04-27:
as you can see the code above, i don't think i'm mixing it.

jeromecoutant, 2021-04-27:
MBED files are usually cpp files, whereas all STM files are c only

pilotak, 2021-04-27:
ok, so what can i do in order to fix it? it's working in F412, F303 and F091 but not on G474

MubeenHCLite, 2021-06-15:
@pilotak 

The issue was reproduced and a PR with required changes is been raised to resolve the issue.
https://github.com/ARMmbed/mbed-os/pull/14778

pilotak, 2021-06-16:
I can confirm the PR solves this issue, perfect. Thank you

closed 2021-06-25 by pilotak