ARMmbed/mbed-os #14531: Print() and (Un)BufferedSerial do not work on Artemis Thing Plus when compiled with ARMC 6.15

idea--list, 2021-04-12:
### Description of defect

Mbed Studio V1.4 updated ARMC to V6.15. Print() and (Un)BufferedSerial do not work on Artemis Thing Plus when compiled with ARMC 6.15. In previous Studio releases these commands worked fine when compiled with ARMC V6.14. Did not try all the Mbed OS API, so am not sure what else might be also affected.

#### Target(s) affected by this defect ?
Artemis Thing Plus (probably all the other Ambiq3 based boards might be affected)

#### Toolchain(s) (name and version) displaying this defect ?
ARMC 6.15

#### What version of Mbed-os are you using (tag or sha) ?

6.8 and 6.9

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio V1.4 on a Win10 x64 machine

#### How is this defect reproduced ? 

compile this code for an Artemis board using ARMC 6.15 and then on any other MCU. Also compile the same code for an Artemis board after configuring Mbed Studio to use GCC_ARM as compiler:

```
#include "mbed.h"
#include <cstdio>
#include <cstring>

#define SERIALBAUD 115200

DigitalOut led(LED_BLUE);
UnbufferedSerial pc(USBTX, USBRX, SERIALBAUD);
char myBuffer[20] = {0};

int main() {

  strcpy(myBuffer, "Hello World!\n");

  if (pc.writable() == true)
    pc.write(myBuffer, sizeof(myBuffer));

  while (1) {
    led = !led;
    strcpy(myBuffer, "Wishing you ");
    if (pc.writable() == true)
      pc.write(&myBuffer, sizeof(myBuffer));
    ThisThread::sleep_for(1000ms);

    led = !led;
    strcpy(myBuffer, "a nice day!\n\n");
    if (pc.writable() == true)
      pc.write(&myBuffer[0], sizeof(myBuffer));
    ThisThread::sleep_for(1000ms);
  }
  return 0;
}
```

0xc0170, 2021-04-12:
cc @ARMmbed/team-sparkfun

idea--list, 2021-04-19:
@0xc0170 
It seems what i noticed originally might be just the tip of the iceberg:

- [here](https://forums.mbed.com/t/bug-with-un-bufferedserial-api-in-mbed-studio-v1-4-for-artemis-board-using-armc-compiler/12774/8) the same issue has been reported for a completely different target
- On MAX32630FTHR a part of my code (USBSerial + USBMSD) is broken if compiled with ARMC6.15, but worked without any issues with ARMC6.14 for months and is still OK when compiled with GCC_ARM

The release notes of ARMC does not contain a changelog, so i can not even guess what might be causing these...and all this issue is unclear for me.
Would be super easy to have 2 links on the Mbed Studio page: 1 for the actual and 1 for the previous release.

0xc0170, 2021-04-20:
You can find Release notes on https://developer.arm.com/tools-and-software/embedded/arm-compiler/downloads/version-6 (select 6.15 version and open release notes. There is a section with what has changed - bugfixes, new cores, etc - all there).

@saheerb What version do we have currently in CI, is it 6.15?

idea--list, 2021-09-23:
After upgrading to Mbed Studio 1.4.2 (comes with ARMC6.16) and to Mbed OS 6.15 this bug seems to be solved.

closed 2021-09-23 by idea--list