ARMmbed/mbed-os #14533: CMake: Command line too long on Ninja/ARMClang on Windows

ccli8, 2021-04-12:
### Description of defect

On Ninja/ARMClang, Win10, `mbed-os-example-for-azure` is failed to compile. The cause seems to related to the bugfix below:

https://github.com/ARMmbed/mbed-os/blob/c73413893fb98aaaeda74513c981ac68adc8645d/CMakeLists.txt#L245-L252

However, after the bugfix, a new issue is introduced: command line too long.

#### Target(s) affected by this defect ?

DISCO_L475VG_IOT01A

#### Toolchain(s) (name and version) displaying this defect ?

ARM Compiler 6.12

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

**mbed-tools**: 7.8.0
**CMake**: 3.19.3
**Ninja**:: 1.10.2

#### How is this defect reproduced ? 

1. Clone `mbed-os-example-for-azure` and navigate to `mbed-os-6.9.0` tag.
    ```sh
    git clone https://github.com/ARMmbed/mbed-os-example-for-azure
    mbed update -C mbed-os-6.9.0
    ```

1. No need to change azure cloud related setting. Just focus on compile process.

1. Patch `CMakeLists.txt` to be compatible with mbed-tools 7.8.0
    <pre>
    set(MBED_CONFIG_PATH ${CMAKE_CURRENT_BINARY_DIR} CACHE INTERNAL "")
    ......
    <s>mbed_set_mbed_target_linker_script(${APP_TARGET})</s>
    </pre>

1. Build on Ninja/ARMClang
    ```
    mbed-tools configure -m DISCO_L475VG_IOT01A -t ARM
    cmake -S . -B cmake_build/DISCO_L475VG_IOT01A/develop/ARM -GNinja -DCMAKE_BUILD_TYPE=develop
    cmake --build cmake_build/DISCO_L475VG_IOT01A/develop/ARM
    ```

1. Error log with command line stripped
    ```
    CreateProcess failed. Command attempted:
    "C:\Keil_v5\ARM\ARMCLANG\bin\armclang.exe --target=arm-arm-none-eabi -DARM_MATH_CM4 ......
    ninja: fatal: CreateProcess: The parameter is incorrect.    
    (is the command line too long?)
    ```

0xc0170, 2021-04-12:
We can't use response files for ARMClang on windows due to the bug there 😢
This is unfortunate and we might need to look at fixing CMake issue (it's still opened as I can see).

0xc0170, 2021-04-20:
Adding the CMake issue here https://gitlab.kitware.com/cmake/cmake/-/issues/21093

0xc0170, 2021-09-03:
The pull request https://gitlab.kitware.com/cmake/cmake/-/merge_requests/6484 was merged, It will be part of the next release.

We can remove the workaround once its fixed. meanwhile, feel free to use their release candidate.

closed 2022-02-14 by 0xc0170