ARMmbed/mbed-os #14535: Serial outputs of mbed-os-example-crash-reporting trimmed in CI

LDong-Arm, 2021-04-13:
### Description of defect

In our CI, [mbed-os-ci_cmake-example-test](https://jenkins-mbedos.oss.arm.com/blue/organizations/jenkins/mbed-os-ci_cmake-example-test/activity/) frequently fails, due to part of the serial output being trimmed/missing. The first [expected](https://github.com/ARMmbed/mbed-os-example-crash-reporting/blob/db45ba408820e8a92e201c7494eb4193480afc09/tests/crash-reporting.log#L1) line is
```
= System will be rebooted due to a fatal error =
```
but in CI misses the serial output of this line, and only starts to have output sometime after that. So the outputs often match the expected logs and the test fails.

Note: This happens when using RaaS. There's no issue on a local board where the entire log, even all the lines before the first line we explicitly check, is intact.

#### Target(s) affected by this defect ?

K64F
(To be tested on more targets)

#### Toolchain(s) (name and version) displaying this defect ?

Any

#### What version of Mbed-os are you using (tag or sha) ?

35b8e558e9eb9b71bc525dcacd6b0df353bdf00a

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-tools 7.8.0

#### How is this defect reproduced ? 

Check failed CI runs in https://jenkins-mbedos.oss.arm.com/blue/organizations/jenkins/mbed-os-ci_cmake-example-test/activity/ or directly run locally-compiled tests on a remote board in RaaS. Both requires access to internal infrastructure.

0xc0170, 2021-04-13:
The first failure I was able to track back, it was 5 days ago. 

The job was mbed-os-ci_cmake-example-test/detail/mbed-os-ci_cmake-example-test/532/pipeline. Prior that green jobs mostly (related to STM32 targets, ble  - nothing generic or k64f).

0xc0170, 2021-04-13:
cmake-test was added also to nightly but I do not see it failing there (but only 2 jobs run with it so not much to compare).

PR #14541 (2021-04-14): Increase baud rate of mbed-os-example-crash-reporting to 115200

### Summary of changes 

At 9600 baud, a large part of the serial output of the crash reporting example is missing on CI boards, causing the example test to fail. At a higher serial speed (115200 baud), the entire output is intact.

This is in alignment with ARMmbed/mbed-os-example-crash-reporting#65.

Fixes #14535

#### Impact of changes
commit b5893beb (2021-04-14): Increase baud rate of mbed-os-example-crash-reporting to 115200

At 9600 baud, a large part of the serial output of the crash reporting
example is missing on CI boards, causing the example test to fail. At
a higher serial speed (115200 baud), the entire output is intact.

This is in alignment with ARMmbed/mbed-os-example-crash-reporting#65.

Fixes #14535
closed 2021-04-14 by 0xc0170