ARMmbed/mbed-os #14544: EFM32 doesn't support QSPI in deep sleep, but there are no locks

amq, 2021-04-14:
### Description of defect

According to the [reference manual of EFM32GG11](https://www.silabs.com/documents/public/reference-manuals/EFM32GG11-RM.pdf), QSPI is not available in EM2 (deep sleep), but I don't see any locks preventing deep sleep in mbed-os.

[image]

Under light load, everything is mostly fine, most of the time. But when I tried a stress test, there are many problems.

#### Target(s) affected by this defect ?

EFM32GG11

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

Any

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Any

#### How is this defect reproduced ?

0xc0170, 2021-04-14:
@ARMmbed/team-silabs Please review

amq, 2021-04-14:
@ciarmcom updated

0xc0170, 2021-04-20:
> Under light load, everything is mostly fine, most of the time. But when I tried a stress test, there are many problems.

If you add locks, all works as expected? Shall they be added?

golgor, 2021-04-21:
> > Under light load, everything is mostly fine, most of the time. But when I tried a stress test, there are many problems.
> 
> If you add locks, all works as expected? Shall they be added?

I was also involved in this issue together with @amq and some other colleagues. We finished some tests today. Our modification in this case was to disable deep sleep before any initialization of the drivers QSP-drivers, effectively forcing you to do dynamic allocation if you don't want to make changes in the drivers yourself.

We have 7 devices with this problem and adding locks fixes the issue for all these 7.

Our original setup:
```c
// FLASH INIT HERE
int main() {
    // FLASH READING/WRITING HERE
    mbed_start_application(POST_APPLICATION_ADDR);
}
```

Our testing setup:
```c
int main() {
    sleep_manager_lock_deep_sleep();
    // FLASH INIT HERE
    // FLASH READING/WRITING HERE
    sleep_manager_unlock_deep_sleep();
    mbed_start_application(POST_APPLICATION_ADDR);
}
```

reiniercoetzer, 2021-05-12:
@ARMmbed/team-silabs @ciarmcom @0xc0170 

Hi guys, any news on this issue?

0xc0170, 2021-05-12:
@ARMmbed/team-silabs Can you have a look?