ARMmbed/mbed-os #14558: BLE Bug: Assert fails on scan when BLE thread is higher priority than initiating thread

tylerhartwig, 2021-04-16:
### Description of defect

An assert in the BLE stack fails when attempting to initiate a scan from a thread that is lower priority than the thread processing BLE events. The assert is as follows:

```
++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: _scan_state == ScanState::pending_scan
Location: 0x4DA8D
File: ./mbed-os/connectivity/FEATURE_BLE/source/generic/GapImpl.cpp+1296
Error Value: 0x0
Current Thread: application_unnamed_thread Id: 0x200042F8 Entry: 0x4EB9D StackSize: 0x1000 StackMem: 0x20019778 SP: 0x2001A5A4 
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&osver=60900&core=0x410FC241&comp=2&ver=90200&tgt=ARDUINO_NANO33BLE
-- MbedOS Error Info --
```

This is meant to be set to pending scan in `initiate_scan` from `GapImpl.cpp`

```
ble_error_t Gap::initiate_scan()
{
    tr_info("Initiate scan");
    const address_t *address = get_random_address(controller_operation_t::scanning);
    if (!address) {
        tr_error("could not get random address");
        return BLE_ERROR_INVALID_STATE;
    }

#if BLE_FEATURE_EXTENDED_ADVERTISING
    if (is_extended_advertising_available()) {
        // set the correct mac address before starting scanning.
        if (_scan_state == ScanState::idle) {
            _pal_gap.set_random_address(*address);
        }

        ble_error_t err = _pal_gap.extended_scan_enable(
            /* enable */true,
            _scan_requested_filtering,
            _scan_requested_duration.value(),
            _scan_requested_period.value()
        );

        if (err) {
            return err;
        }
    } else
#endif // BLE_FEATURE_EXTENDED_ADVERTISING
    {
        if (_scan_requested_period.value() != 0) {
            tr_error("scan request period cannot be equal to 0");
            return BLE_ERROR_INVALID_PARAM;
        }

        // update the address if no scan or advertising is running
        auto adv_handle = LEGACY_ADVERTISING_HANDLE;
        if ((_scan_state == ScanState::idle) && !_active_sets.get(adv_handle) && !_pending_sets.get(adv_handle)) {
            _pal_gap.set_random_address(*address);
        }

        ble_error_t err = _pal_gap.scan_enable(
            true,
            _scan_requested_filtering == duplicates_filter_t::DISABLE ? false : true
        );

        if (err) {
            tr_error("initiate failed in regular advertising");
            return err;
        }

        _scan_timeout.detach();
        if (_scan_requested_duration.value()) {
            _scan_timeout.attach(
                [this]() { _event_queue.post([this] { process_legacy_scan_timeout(); }); },
                _scan_requested_duration.valueChrono()
            );
        }
    }

    tr_info("About to set scan state");
    set_scan_state(ScanState::pending_scan);

    return BLE_ERROR_NONE;
}
```

The lines previous to `set_scan_state(ScanState::pending_scan)` initiate the scan to the hardware, resulting in the following function being called in the thread handling BLE events.

```

#if BLE_ROLE_OBSERVER
void Gap::on_scan_started(bool success)
{
    tr_info("Scan %s", success ? "successfully started" : "failed to start");

    MBED_ASSERT(_scan_state == ScanState::pending_scan);

    if (success) {
        set_scan_state(ScanState::scan);
        /* if no longer want the scan */
        if (!_scan_requested) {
            stopScan();
        }
    } else {
        set_scan_state(ScanState::idle);
    }
}
```

At this point, `_scan_state` is not `pending_scan` as expected, and the assert is triggered. 

#### Target(s) affected by this defect ?

Arduino Nano 33 BLE is affected, and I assume all other targets would be affected.

#### Toolchain(s) (name and version) displaying this defect ?

mbed os 6.9.0 (I believe this started prior to 6.9.0 however)

#### What version of Mbed-os are you using (tag or sha) ?

mbed os 6.9.0 -- tested on `master` as well.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A -- Can add additional details if needed.

#### How is this defect reproduced ? 

Minimal reproduction code:

```
#include "mbed.h"
#include "mbed_trace.h"

#define TRACE_GROUP "main"

static rtos::Thread ble_thread(osPriorityAboveNormal, OS_STACK_SIZE);
static rtos::Thread scan_thread(osPriorityNormal, OS_STACK_SIZE);
static events::EventQueue event_queue(32 * EVENTS_EVENT_SIZE);

void schedule_ble_events(BLE::OnEventsToProcessCallbackContext* event){
    event_queue.call(&event->ble, &BLE::processEvents);
}

class BLEBug : public ble::Gap::EventHandler {
public:
    BLE* ble_interface;
    ble::Gap* gap;
    bool wait_for_scan = true;

    void on_init_complete(BLE::InitializationCompleteCallbackContext* event) {
        printf("Init Complete\n");
        wait_for_scan = false;
    }

    void start_scan(){
        while(wait_for_scan) {
            rtos::ThisThread::sleep_for(1000);
        }

        printf("Starting Scan\n");
        ble::ScanParameters scanParameters;
        gap->setScanParameters(scanParameters);
        gap->startScan();
    }

    void init() {
        ble_interface = &BLE::Instance();
        gap = &ble_interface->gap();

        gap->setEventHandler(this);

        ble_interface->onEventsToProcess(&schedule_ble_events);

        ble_interface->init(this, &BLEBug::on_init_complete);
        ble_thread.start(mbed::callback(&event_queue, &events::EventQueue::dispatch_forever));
        scan_thread.start(mbed::callback(this, &BLEBug::start_scan));
    }

};

BLEBug bug;

int main(){
    mbed_trace_init();
    bug.init();
}
```

I think I've narrowed in on the bug fairly closely, I can help fix it as well -- though I'm not sure which approach is best to take here. I have some questions regarding this as follows:

- Is it meant for these functions to be called in separate threads as I have done?
- Adjusting the ASSERT fixes the problem, however could result in the wrong `_scan_state` being set
- Conditional logic can be added to `on_scan_started` and `initiate_scan` to check the state before setting, as appropriate, however this adds additional complexity 
- Call ordering can be adjusted so `pending_scan` is set before initiating the scan.
- The initiation of the scan and setting of the state could be done in a critical section to ensure appropriate thread safety.

Looking for guidance on what best aligns with the rest of the mbed os systems, I don't mind implementing and creating the PR :).

0xc0170, 2021-04-20:
@ARMmbed/mbed-os-connectivity please review

pan-, 2021-04-21:
> Is it meant for these functions to be called in separate threads as I have done?

No, Bluetooth APIs are not thread safe, it is not meant to be called from various threads. The best way to communicate between threads is with event queues. It is fairly easy to do ow that Mbed OS supports C++11. The lambda notation is really powerfull: 

```c++
    void start_scan(){
        while(wait_for_scan) {
            rtos::ThisThread::sleep_for(1000);
        }

        printf("Starting Scan\n");
        // Execute this by the event queue processing BLE. 
        event_queue.call([this] { 
                ble::ScanParameters scanParameters;
                gap->setScanParameters(scanParameters);
                gap->startScan();
        });
    }
```

closed 2021-04-21 by pan-

tylerhartwig, 2021-04-21:
@pan- thanks! this makes sense -- is this documented anywhere? That wasn't immediately apparent to me from the docs, but I easily could have missed something.

paul-szczepanek-arm, 2021-04-21:
It only mentions that it depends how you handle the ble.onEventsToProcess function and your API calls but BLE API itself does not enforce any thread safety. We should put some clear guidance in the BLE header.