ARMmbed/mbed-os #14559: A suspected bug in BLE bonding

16L-YT, 2021-04-18:
### Description of defect

When I try to add simple bonding feature to my BLE device by adapting most parts of [SecurityAndPrivacy] (https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_SecurityAndPrivacy/source) example , and after getting rid of `SecurityCentral` class, as my case includes only a peripheral (my ble device) and an Android phone trying to bond with it. When I try to connect to the device using NRF connect, I have a popup from Android to pair with device 2 times in sequence. After all, the bonding fail and Android shows me that bonding is failed due to incorrect PIN or passphrase although I don't use it in my application. I tried 2 different Android phones to make sure it is not an issue related to my phone. I also tried to add passphrase and I got the same result. I have a fealing that this is due to a miss-leading documentation of SecurityManager calss or to a bug in Mbed somewhere. 

This is the log from NRF Connect while connection: 

```
nRF Connect, 2021-04-18
BLEDevSma (06:05:04:92:95:07)
V	17:57:00.661	Connecting to 06:05:04:92:95:07...
D	17:57:00.661	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, preferred PHY = LE 1M)
D	17:57:03.106	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_BONDING (11)
D	17:57:03.120	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	17:57:03.120	Connected to 06:05:04:92:95:07
D	17:57:03.191	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	17:57:03.191	[Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_CONSENT (3)
I	17:57:03.526	Connection parameters updated (interval: 7.5ms, latency: 0, timeout: 5000ms)
I	17:57:04.396	Connection parameters updated (interval: 48.75ms, latency: 0, timeout: 5000ms)
D	17:57:06.990	[Broadcast] Action received: android.bluetooth.device.action.PAIRING_REQUEST, pairing variant: PAIRING_VARIANT_CONSENT (3)
D	17:57:08.851	[Broadcast] Action received: android.bluetooth.device.action.BOND_STATE_CHANGED, bond state changed to: BOND_NONE (10)
I	17:57:08.851	Bonding failed
V	17:57:08.893	Discovering services...
D	17:57:08.893	gatt.discoverServices()
D	17:57:08.965	[Callback] Services discovered with status: 0
I	17:57:08.965	Services discovered
V	17:57:08.994	Generic Access (0x1800)

```
[image]
[image]

Here is the code I run: 

```
/*
* adapted from: mbed-os-ble-utils/ble_process.h &
* mbed-os-example-ble/BLE_SecurityAndPrivacy
*/

#ifndef DEV_ADV_H_
#define DEV_ADV_H_

#include <stdint.h>
#include <string>
#include <cstring>

#include <events/mbed_events.h>
#include "ble/BLE.h"

static const std::chrono::milliseconds delay = 3000ms;

class DevAdvSecurity : private mbed::NonCopyable<DevAdvSecurity>,
                     public SecurityManager::EventHandler,
                     public ble::Gap::EventHandler
{
public:
    DevAdvSecurity(BLE &ble, events::EventQueue &event_queue) :
        _ble(ble), _event_queue(event_queue),_post_init_cb()  { };

    virtual ~DevAdvSecurity()
    {
        _ble.onEventsToProcess(nullptr);
    };

    /** Start BLE interface initialisation */
    void run()
    {
        /* this will inform us off all events so we can schedule their handling
         * using our event queue */
        _ble.onEventsToProcess(makeFunctionPointer(this, &DevAdvSecurity::schedule_ble_events));
[... 282 lines cut ...]
    string name = "BLEDevSma";
    dev_preipheral.set_dev_name(name);

    Device services;

    dev_preipheral.register_oninit_cb(callback(&services, &Device::start));
    
    dev_preipheral.run();
```
 

#### Target(s) affected by this defect ?
AMA3B1KK-KBR-B0

#### Toolchain(s) (name and version) displaying this defect ?
gcc version 10.2.1 20201103 (release) (xPack GNU Arm Embedded GCC, 64-bit)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Tools that comes with Mbed Studio 1.4.0

#### How is this defect reproduced ?

0xc0170, 2021-04-20:
@ARMmbed/mbed-os-connectivity Please review

16L-YT, 2021-04-22:
@0xc0170 I know that there is no any obligation to check the bug (if exists) during a constrained time. But is there any time frame for mbed-os-connectivity team to check the issue as It is blocking further development from our side.

16L-YT, 2021-04-25:
UPDATE: 

I can ask a more precise question: According to #7300 

> Unfortunately writing in the flash while BLE is on was out of the scope for the release; it is a complex process as when the flash is written.

According to my understanding, mbed-os can't handle storing the Bonding information to Flash without a filesystem already initialized. So a call to  `securityManager` like this will always produce a fail in Bonding. 

```
        error = _ble.securityManager().init(
            /* enableBonding */ true,
            /* requireMITM */ false,
            /* iocaps */ SecurityManager::IO_CAPS_NONE,
            /* passkey */ nullptr,
            /* signing */ false,
            /* dbFilepath */ nullptr
        );
```

And according to [this](https://devzone.nordicsemi.com/f/nordic-q-a/43338/android-pairing-without-bonding---nrf-connect) there is no way to ask Android to do pairing without bonding. 

As a result and as Ambiq has no `LOCALFILESYSTEM` feature, then actually bonding can't be done. 

Is my understanding correct ? Any workaround if it is correct ?

paul-szczepanek-arm, 2021-04-26:
Sorry, I looked at this on Friday but it appears something went wrong posting it :( Let me try again:
I need the rest of your code to figure out what is happening. Maybe you can make a repo that contains your code and I can check it out. The current code could be disconnecting after 2 seconds (look at the _bonded variable). Example code is just for educational purposes it's not necessarily a good base for real code (we are working on something like that).
As for bonding without a FS, it does work. The bonding information is kept in RAM. Works the same but obviously will be lost on reset.

kalle16lab, 2021-04-28:
I've used the the https://github.com/ARMmbed/mbed-os-example-ble/tree/development/BLE_SecurityAndPrivacy example without modification on an Ambiq target and am also unable to pair the device with an android phone. The pairing request always fails with error code " SEC_STATUS_CONFIRM_VALUE        = 0x84,  /**< Confirm value failed. */"

paul-szczepanek-arm, 2021-04-28:
I don't have an Ambiq board. What LL firwmare is on the BLE chip?

paul-szczepanek-arm, 2021-04-28:
The reason I ask is because I remember I had to make a workaround for the wb55 firmware for a bug in the LL that was causing the same kind of issue you're describing.

16L-YT, 2021-04-29:
The same one the latest mbed has:

```
/*! \brief  Version number. */
#define LL_VER_NUM          1366
```

How to compare with cordio used with Ambiq sdk ? Because bonding seems working using the official sdk without mbed os .

The following is from wsf/include/ll_defs.h from AmbiqSuite-R2.5.1
```
 *  \brief      Link layer constant definitions.
 *
 *  $Date: 2017-03-10 20:10:06 -0600 (Fri, 10 Mar 2017) $
 *  $Revision: 11511 $
```

Also mbed is using Ambiq ble related source files from 2.4.2 version. An update to 2.5.1 version did not solved the issue (please check the following 2 commits).

https://github.com/16Lab-Inc/mbed-os/commit/02c92998e8fc46b7ea2ed3443ab6efc1fc79f14a
https://github.com/16Lab-Inc/mbed-os/commit/aad39c01e7a38320b2579b07e51d27aacbe2731b

kalle16lab, 2021-04-29:
[Here](https://github.com/ARMmbed/mbed-os/tree/master/connectivity/drivers/ble/FEATURE_BLE/COMPONENT_BlueNRG_2) under "Known limitations" it says that security does not work due to pairing failure. Has pairing been confirmed to work on any targets yet? It seems like this issue is not Ambiq target specific.

paul-szczepanek-arm, 2021-04-29:
It works on nrf52840 and DISCO_L475VG_IOT01A and the bluenrg 1 shield.  The bluenrg 2 indeed had some pairing issues - hence the note.

pan-, 2021-04-29:
@16L-YT Can you run your application with traces and share the traces with us. This document: https://os.mbed.com/docs/mbed-os/v6.9/apis/ble.html explain how to enable traces.

paul-szczepanek-arm, 2021-04-29:
If the CONFIRM_VALUE is wrong that means that the local calculation of the value doesn't match the one done on the remote. This is most likely caused by disagreement on addresses - one of the values used in the calculation  of the confirm value is the address of the device. This is the most likely point where a bug could exist so try to verify the values used for the confirm value calculation are correct.

kalle16lab, 2021-04-29:
With mbed_app.json like this:
```
{
    "target_overrides": {
        "*": {
            "mbed-trace.enable": true,
            "mbed-trace.max-level": "TRACE_LEVEL_DEBUG",
            "cordio.trace-hci-packets": false,
            "cordio.trace-cordio-wsf-traces": true,
            "ble.trace-human-readable-enums": true,
            "target.macros_add": ["MBED_MINIMAL_PRINTF"],
            "target.console-uart-flow-control": null,
            "ble.ble-gap-host-based-private-address-resolution": false,
            "cordio.rpa-swap-workaround": true,
            "cordio.trace-pal-echoes": 1,
            "platform.stdio-baud-rate": 115200
        }
    }
}
```
I get the following trace:
[trace_pair.txt](https://github.com/ARMmbed/mbed-os/files/6399387/trace_pair.txt)

NB! setting "cordio.trace-hci-packets": true, hard crashes the device.

Intresting things I see:
* `[INFO][BLGP]: Connection 1 successfully completed - role=PERIPHERAL, peer_address_type=RANDOM peer_address=d1:5f:31:53:7b:5a, connection_interval=40, connection_latency=0, supervision_timeout=500, local_resolvable_private_address=00:00:00:00:00:00, peer_resolvable_private_address=00:00:00:00:00:00`

local address is 0

* `[INFO][BLSM]: SM init: bondable=false, mitm=false, iocaps=IO_CAPS_NONE, passkey=0x45DD8, signing=false, db_path=0x386B8`

passkey is not 0 even though init is called like this:
```
        error = _ble.securityManager().init(
            /* enableBonding */ false,
            /* requireMITM */ false,
            /* iocaps */ SecurityManager::IO_CAPS_NONE,
            /* passkey */ nullptr,
            /* signing */ false,
            /* dbFilepath */ db_path
        );
```

* `[INFO][BLSM]: Set stack display passkey: 30:30:30:30:30:30`

passkey is set even though a null pointer is passed to securityManager().init()

* `[ERR ][BLSM]: Host based private address resolution enabled but not supported by the controller`

even though  `"ble.ble-gap-host-based-private-address-resolution": false,` is defined in mbed_app.json

paul-szczepanek-arm, 2021-04-29:
`[ERR ][BLSM]: Host based private address resolution enabled but not supported by the controller` that error message appears to be misleading,  I think it should say: `disabled` not `enabled`. Since your controller doesn't support privacy you need to enable "ble.ble-gap-host-based-private-address-resolution" if you want to use privacy.

paul-szczepanek-arm, 2021-04-29:
As for the ```SM init: bondable=false, mitm=false, iocaps=IO_CAPS_NONE, passkey=0x45DD8, signing=false, db_path=0x386B8``` showing the passykey as not null, the line with the trace print has an error:
`tr_info("SM init: bondable=%s, mitm=%s, iocaps=%s, passkey=%p, signing=%s, db_path=%p",`
should be
`tr_info("SM init: bondable=%s, mitm=%s, iocaps=%s, passkey=%s, signing=%s, db_path=%p",`
so it prints the pointer instead of the string.
Thanks, I'll fix that along with the error message.

kalle16lab, 2021-04-29:
Even with this mistake the printed value should still be 0x00. Because a null pointer should be passed to it or is the nullptr passed to _ble.securityManager().init() transformed into a string before printing?

As far as I can tell the printing happens directly after the init call or am I missing something? :
```ble_error_t SecurityManager::init(
    bool bondable,
    bool mitm,
    SecurityIOCapabilities_t iocaps,
    const uint8_t* passkey,
    bool signing,
    const char* db_path
)
{
    tr_info("SM init: bondable=%s, mitm=%s, iocaps=%s, passkey=%p, signing=%s, db_path=%p",
        to_string(bondable),
        to_string(mitm),
        to_string(iocaps),
        passkey_str(passkey),
        to_string(signing),
        db_path
    );
```
Also later the passkey is printed as `30:30:30:30:30:30`. Is this expected behaviour? as far as I can tell having the local_resolvable_private_address be 00:00:00:00:00:00 is not valid.

with "ble.ble-gap-host-based-private-address-resolution": true
```[INFO][BLCO]: WSF Cordio tracing enabled
[INFO][BLE ]: Allocated 9000 bytes for Cordio
[INFO][BLE ]: BLE features enabled: EXTENDED_ADVERTISING GATT_CLIENT GATT_SERVER PERIODIC_ADVERTISING PHY_MANAGEMENT PRIVACY SECURE_CONNECTIONS SECURITY SIGNING WHITELIST
[INFO][BLE ]: BLE roles enabled: PERIPHERAL CENTRAL
[INFO][BLE ]: Initialising Cordio host stack
[DBG ][BLE ]: HCI handler id: 0
[DBG ][BLE ]: DM handler id: 1
[DBG ][BLE ]: L2C slave handler id: 2
[DBG ][BLE ]: ATT handler id: 3
[DBG ][BLE ]: SMP handler id: 4
[DBG ][BLE ]: stack handler id: 5

 * Device is a peripheral *

[INFO][BLPR]: Initialize privacy PAL
[INFO][BLE ]: Initialising BLE instance
[INFO][BLHC]: CordioHCIDriver initializing
[DBG ][BLE ]: Cordio HCI reset started
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_RESET
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_SET_EVENT_MASK
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_SET_EVENT_MASK
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_SET_EVENT_MASK_PAGE2
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_READ_BD_ADDR
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_READ_BUF_SIZE
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_READ_SUP_STATES
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_READ_WHITE_LIST_SIZE
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_READ_LOCAL_SUP_FEAT
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_READ_MAX_DATA_LEN
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_WRITE_DEF_DATA_LEN
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_RAND
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_RAND
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_RAND
[DBG ][BLE ]: Reset sequence command complete: HCI_OPCODE_LE_RAND
[DBG ][BLHC]: Signal cordio HCI reset sequence done
[DBG ][BLE ]: stack_handler received DM_RESET_CMPL_IND
[INFO][BLE ]: Reset sequence complete. Controller supported features:
[INFO][BLE ]: ENCRYPTION
[INFO][BLE ]: DATA_LEN_EXT
[INFO][BLE ]: Initialising extended features
[INFO][BLGS]: Initialize GattServer
[INFO][BLGS]: Add Generic Access Service to the Gatt Server: handle=1
[INFO][BLGS]: Add Device Name characteristic: handle=3
[INFO][BLGS]: Add Appearance characteristic: handle=5
[INFO][BLGS]: Add Peripheral Preferred Connection characteristic: handle=7
[INFO][BLGS]: Add Generic Attribute Service to the Gatt Server: handle=8
[INFO][BLGS]: Add Service Changed characteristic: handle=10
[... 146 lines cut ...]
[INFO][BLSM]: Connection 1 - MITM performed: false
Pairing failed with error code: 84
[DBG ][BLCO]: smpDbGetRecord: connId: 1 type: 1
[DBG ][BLCO]: smpDbAddDevice
[DBG ][BLCO]: SmpDbMaxAttemptReached: connId: 1
[DBG ][BLCO]: smpSmExecute event=16 state=38
[INFO][BLGP]: Connection 1: disconnect - reason=USER_TERMINATION
[INFO][BLDM]: Connection 1: disconnect - disconnection_reason=USER_TERMINATION
[DBG ][BLCO]: dmConnSmExecute event=25 state=3
[DBG ][BLCO]: dmConnSmExecute event=29 state=4
[DBG ][BLCO]: dmConnCcbDealloc 1
[DBG ][BLE ]: Connection 1 - Clear CCC table
[DBG ][BLCO]: AttsCccClearTable connId=1
[DBG ][BLCO]: smpDbGetRecord: connId: 1 type: 1
[DBG ][BLCO]: SmpDbSetFailureCount: connId: 1 count: 0
[DBG ][BLCO]: smpSmExecute event=10 state=0
[DBG ][BLE ]: stack_handler received DM_CONN_CLOSE_IND
[DBG ][BLGP]: GAP event of type DISCONNECTION_COMPLETE received
[INFO][BLGP]: Disconnection 1 successfully completed - reason=19
[INFO][BLSM]: Connection 1 - Process disconnection
[INFO][BLSM]: Connection 1 - Remove stack peer CSRK
Failed to bond.

 * Device is a central *
```

paul-szczepanek-arm, 2021-04-29:
Yeah, it's converted to a string. Is this trace the trace of an unchanged example?

kalle16lab, 2021-04-29:
Basically unmodified.
```
    mbed_trace_init();
    mbed_trace_config_set(TRACE_ACTIVE_LEVEL_ALL);
```
has been added to main();
also 
```
        error = _ble.securityManager().init(
            /* enableBonding */ false,
            /* requireMITM */ false,
            /* iocaps */ SecurityManager::IO_CAPS_NONE,
            /* passkey */ nullptr,
            /* signing */ false,
            /* dbFilepath */ db_path
        );
```
enableBonding has been set to false because we suspected the lack of file-system is to blame.

paul-szczepanek-arm, 2021-04-29:
Just to exclude problems with android could you please run the example on two of your ambiq boards?

kalle16lab, 2021-04-29:
Different colors are different Boards. Also no more specific traces since they all hard crash. Mostly with file accessed in ISR error.
[image]

paul-szczepanek-arm, 2021-04-30:
OK, I need to get an artemis board. This board is not yet officially supported and obviously we never tested ble on it. I'll get back to you when I get the board but I cannot guarantee this will be fixed.

kalle16lab, 2021-04-30:
Doing some digging I found that in the HCITransportDriver setting a random address is supressed. Would this cause pairing not to work?
```
uint16_t AP3CordioHCITransportDriver::write(uint8_t packet_type, uint16_t len, uint8_t *data)
{
#if PRINT_DEBUG_HCI
    printf("sent tx packet_type: %02X data: ", packet_type);
    for (int i = 0; i < len; i++)
    {
        printf(" %02X", data[i]);
    }
    printf("\r\n");
#endif
    //Temporary workaround, random address not working, suppress it.
    if (data[0] == 0x06 && data[1] == 0x20)
    {
#if PRINT_DEBUG_HCI
        printf("LE Set Advertising Params\r\n");
#endif
        data[8] = 0;
    }
```

paul-szczepanek-arm, 2021-04-30:
Yes, if addresses don't match between the devices the confirm value will be different and therefore pairing will fail.

kalle16lab, 2021-05-03:
I've removed the following part from `AP3CordioHCITransportDriver::write`
```
  //Temporary workaround, random address not working, suppress it.
  if (data[0] == 0x06 && data[1] == 0x20)
  {
#if PRINT_DEBUG_HCI
      printf("LE Set Advertising Params\r\n");
#endif
      data[8] = 0;
  }
```
and this breaks BLE. 

Any ideas on why this happens? What does the command being suppressed specifically do? What would be the workaround if the controller doesn't support the command being suppressed? Could this command not working be a CORDIO stack issue?

paul-szczepanek-arm, 2021-05-05:
Judging by the comment in the code setting random address breaks the controller so that line stops you from setting the address type to random when advertising. The code checks if the HCI command is "LE Set Advertising Parameters command" (page 2482 in the 5.2 ble spec), the 6th byte that it overwrites (data[8]) is the "Own_Address_Type" and 0 is the public address. In effect this means the board will always advertise with public address and you cannot use privacy as it will cause the confirm values to differ since each side thinks the address is different.

kalle16lab, 2021-05-05:
Is pairing/bonding possible without supporting a random address during advertising? I expect setting a random address manually could be possibility.

kalle16lab, 2021-05-05:
Also the example code provided by the vendor performs paring/bonding without issue. I was surprised to find that mbed-os's SecurityManager is not based on cordio's SMP. Are these incompatible or is there a way to use the entire cordio stack with mbed-os?

paul-szczepanek-arm, 2021-05-05:
You can probably pair and bond but only with a public address - so you cannot enable privacy (which requires random addresses).

pan-, 2021-05-05:
> Is pairing/bonding possible without supporting a random address during advertising? I expect setting a random address manually could be possible workaround.

Yes it is possible to use public addresses to pair and bond however if you cannot set a random address in the controller then the privacy won't work. When privacy is enabled, your device advertise with a private resolvable address which is renewed every 15 minutes. Until Bluetooth 5, this private resolvable address was computed on the host and set in the controller using the set random address HCI command. With Bluetooth 5, link layer privacy has been introduced and the controller can generate an address when it advertise, scan or initiate connection. Not all BT5 controller supports this feature. 

To solve your issue I suggest you remove privacy related bits and use a public address: 

```c++
class DevAdvSecurity : private mbed::NonCopyable<DevAdvSecurity>,
                     public SecurityManager::EventHandler,
                     public ble::Gap::EventHandler
{
public:
    DevAdvSecurity(BLE &ble, events::EventQueue &event_queue) :
        _ble(ble), _event_queue(event_queue),_post_init_cb()  { };

    virtual ~DevAdvSecurity()
    {
        _ble.onEventsToProcess(nullptr);
    };

    /** Start BLE interface initialisation */
    void run()
    {
        /* this will inform us off all events so we can schedule their handling
         * using our event queue */
        _ble.onEventsToProcess(makeFunctionPointer(this, &DevAdvSecurity::schedule_ble_events));

        /* handle gap events */
        _ble.gap().setEventHandler(this);

        if (_ble.hasInitialized()) {
            /* ble instance already initialised, skip init and start activity */
            start();
        } else {
            ble_error_t error = _ble.init(this, &DevAdvSecurity::on_init_complete);

            if (error) {
                return;
            }
        }

        if (_post_init_cb) {
            _post_init_cb(_ble, _event_queue);
        }

        /* this will not return until shutdown */
        _event_queue.dispatch_forever();
    };

    /** Override to start chosen activity when the system starts */
    virtual void start() = 0;

    /** This is called when BLE interface is initialised and starts the demonstration */
    void on_init_complete(BLE::InitializationCompleteCallbackContext *event)
    {
        ble_error_t error;

        if (event->error) {
            return;
        }

        // SecurityManager::Passkey_t passkey = {0,0,0,0,0,0};

        error = _ble.securityManager().init(
            /* enableBonding */ true,
            /* requireMITM */ false,
            /* iocaps */ SecurityManager::IO_CAPS_NONE,
            /* passkey */ nullptr,
            /* signing */ false,
            /* dbFilepath */ nullptr
[... 178 lines cut ...]
            ble::adv_interval_t(ble::millisecond_t(1000)),
             ble::adv_interval_t(ble::millisecond_t(2000))
        );

        adv_parameters.setOwnAddressType(ble::own_address_type_t::PUBLIC);

        // adv_params.setTxPower(-127);

        error = _ble.gap().setAdvertisingParameters(ble::LEGACY_ADVERTISING_HANDLE, adv_parameters);

        if (error) {
            return;
        }

        error = _ble.gap().startAdvertising(ble::LEGACY_ADVERTISING_HANDLE);

        if (error) {
            return;
        }
    }

protected:
    char adv_name[32];
};
```

kalle16lab, 2021-05-06:
Thank you this worked. I'm now able to pair with an Android device but for some reason pairing confirmation on Android side always has to be preformed twice. The vendor provided cordio based examples succeed on the first attempt. It seems like the paring is initiated twice for some reason.

pan-, 2021-05-20:
Follow up in https://github.com/ARMmbed/mbed-os/issues/14654

closed 2021-05-20 by pan-