ARMmbed/mbed-os #14563: WiSUN BR can not forward multicast from host to wisun network

mcuxmx, 2021-04-20:
### Description of defect
I built the wisun network.

Add BR and all nodes to group ff05:: 7 through socket, and then host (Windows) sends data to multicast, hoping that BR and nodes can receive data.

1. I add BR's ethernet interface and node's mesh interface to ff05:: 7, send data through host, BR can receive data, but node can't receive data.

2. I add BR's ethernet interface and node's mesh interface to ff05:: 7.  send data through host, but BR and node can't receive data.

3. When I use BR send data to ff05:: 7. BR, node and host can receive data.

#### Target(s) affected by this defect ?
BR: DISCO_F769NI+S2LP radio
NODE: NUCLEO_F429ZI+S2LP radio
#### Toolchain(s) (name and version) displaying this defect ?
any

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
any

#### How is this defect reproduced ? 

alyways

0xc0170, 2021-04-20:
@artokin Please review

mikter, 2021-04-28:
I tested this with the latest Wi-SUN stack and the exact same case is working as expected. I am not sure if there is some fixes that are affecting this case, but using the latest should be tested.

Also to identify the issue could you take wireshark captures from the backbone if we can identify what is preventing the forwarding. One problem might be the HOP count of multicast ping that Windows machine is sending if that is not high enough by default the packet will be dropped before reaching the node.

mcuxmx, 2021-04-28:
@mikter
Thank you for your reply.
Indeed, as you said, through Wireshark monitoring, I found that the TTL issued by my tool was 1. I changed a tool and multicast worked well.

closed 2021-04-28 by mcuxmx