ARMmbed/mbed-os #14577: [BLE]: Indication confirmation never received [AmbiqMicro target]

kalle16lab, 2021-04-22:
### Description of defect

Using the provided [BLE_GattServer_CharacteristicUpdates](https://github.com/ARMmbed/mbed-os-example-ble/tree/development/BLE_GattServer_CharacteristicUpdates) sample code. When enabling indications using nRFConnect app on android the following function is never called:
```
void onConfirmationReceived(const GattConfirmationReceivedCallbackParams &params) override
{
    printf("confirmation received on handle %d\r\n", params.attHandle);
}
```
#### Target(s) affected by this defect ?
This has been observed on AmbiqMicro Apollo 3 Blue target

#### Toolchain(s) (name and version) displaying this defect ?

gcc version 9.3.1 20200408 (release) (GNU Arm Embedded Toolchain 9-2020-q2-update)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio: 1.4.0
#### How is this defect reproduced ? 
Compile the following example for AmbiqMicro target:

    https://github.com/ARMmbed/mbed-os-example-ble/tree/development/BLE_GattServer_CharacteristicUpdates

Enable indications using nRF Connect android app. See that "sent updates" is being printed but "confirmation received on handle %d" is never printed.

0xc0170, 2021-04-22:
@ARMmbed/team-sparkfun Please review

paul-szczepanek-arm, 2021-04-26:
Thank you for spotting this and reporting back to us. I looked into it and this functionality doesn't look to be implemented correctly in mbed-os. We have to clarify the interpretation of events for data sent and confirmation as currently the indications are not producing separate confirmation events. The data sent event is produced by the received confirmation.

paul-szczepanek-arm, 2021-04-26:
Yes, onDataSent is called when indication confirmation is received. I'll give you an ETA tomorrow. We need to decide how to deprecate current behaviour.

paul-szczepanek-arm, 2021-04-27:
Mbed-os and Cordio have taken incompatible design decisions on this. We cannot reconcile the events without changing Cordio and we don't want to diverge from upstream. We are going to align with Cordio on this.

This means the onConfirmationReceived event will be deprecated and documentation updated to explain clarify the current behaviour:
- if there is no confirmation onDataSent is called when packet is sent,
- if there is a confirmation onDataSent is called when confirmation is received.

@kalle16lab I will make a PR this week but since the behaviour will not change from current behaviour (only documentation) you can keep developing using the current version.

paul-szczepanek-arm, 2021-04-28:
deprecation PR https://github.com/ARMmbed/mbed-os/pull/14602

closed 2021-04-28 by paul-szczepanek-arm