ARMmbed/mbed-os #14579: [BLE]: Tx power level setting on AmbiqMicro target

idea--list, 2021-04-22:
### Description of defect

Have been long looking for a method to set BLE Tx power for the Apollo3 MCU. Seems i found the relevant `HciVsA3_SetRfPowerLevelEx` function, however after adding

`#include "mbed-os/connectivity/drivers/ble/FEATURE_BLE/TARGET_Ambiq_Micro/hal/apollo3/hci_drv_apollo3.h"` and `HciVsA3_SetRfPowerLevelEx(TX_POWER_LEVEL_PLUS_3P0_dBm);` to the main.cpp i get unknown type errors inside hci_drv_apollo3.h.

Even uint8_t and uint16_t are reported as unknown types. So i add `#include "mbed.h"` to that file.
However after doing so `bool_t`, `wsfEventMask_t`, `wsfMsgHdr_t` and `wsfHandlerId_t` are still reported to be unknown. Could not find where these types are defined, which renders the `HciVsA3_SetRfPowerLevelEx` function unusable.

#### Target(s) affected by this defect ?
Artemis Thing Plus (probably all Ambiq Apollo 3 Blue based boards are affected)

#### Toolchain(s) (name and version) displaying this defect ?
ARMC 6.15 and GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS 6.9

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio V1.4 (on Win10 x64)

#### How is this defect reproduced ? 

Try to include `hci_drv_apollo3.h` into any example of the [mbed-os-example-ble repo](https://github.com/ARMmbed/mbed-os-example-ble) and then call `HciVsA3_SetRfPowerLevelEx()`. It will throw unknown type errors and will not even compile.

0xc0170, 2021-04-22:
@ARMmbed/team-sparkfun Please review

yahyatawil, 2021-04-23:
@idea--list A side question. Just wondering, why you don't use `setTxPower` method from `AdvertisingParameters` class ?

idea--list, 2021-04-23:
@yahyatawil 
Primarily because i did not know that it exists. It is buried way too deep in the docs. I actually found it right now mentioned [here](https://forums.mbed.com/t/urgently-need-to-limit-ble-advertising-to-one-channel/7899/2). Someone asked about how to set gap parameters in OS 5.11 (vcoubard shared a link there).

After getting know about that i simply can not figure out the proper syntax how to use it here, tried some variations i could think of but none were correct:

```
ble::AdvertisingParameters adv_parameters(
   ble::advertising_type_t::CONNECTABLE_UNDIRECTED,
   ble::adv_interval_t(ble::millisecond_t(1000))
);
```

So i ended up chaining it 5 lines later like that:

```
ble_error_t error = _ble.gap().setAdvertisingParameters(
   ble::LEGACY_ADVERTISING_HANDLE,
   adv_parameters.setTxPower(-10)
);
```
However it does not seem to have any effect on this target: power consumption and received signal strength is the same even if i use `adv_parameters.setTxPower(3)`

Maybe i am doing something wrong, but i am really struggling with Mbed's BLE implementation for several reasons.

pan-, 2021-04-23:
@yahyatawil @idea--list It still make sense to use vendor extensions for controllers that don't support that BT5 feature and/or want to raise the global RF power so it applies to connections too; not just advertising.

16L-YT, 2021-04-24:
@pan- The reason of my question was to know if @idea--list encountered the same case I faced (it seems yes). Where, `setTxPower` almost has no change to the power of advertisement. I have also Apollo3 MCU and I used `setTxPower` like this way:

```
        ble::AdvertisingParameters adv_parameters(
            ble::advertising_type_t::CONNECTABLE_UNDIRECTED,            
            ble::adv_interval_t(ble::millisecond_t(1000)),
             ble::adv_interval_t(ble::millisecond_t(2000))
        );

        adv_params.setTxPower(-127);

        error = _ble.gap().setAdvertisingParameters(ble::LEGACY_ADVERTISING_HANDLE, adv_parameters);
```

The peak current of advertisements packets has almost know change when I played with the 'advertising_power_t txPower' value. 

[image]

PR #14585 (2021-04-28): add #include "wsf_os.h" into hci_drv_apollo3.h

### Summary of changes 

Fixes #14579 
After adding `#include "wsf_os.h"` to `hci_drv_apollo3.h` i can place `HciVsA3_SetRfPowerLevelEx(TX_POWER_LEVEL_PLUS_3P0_dBm);` just before `error = _ble.gap().startAdvertising(ble::LEGACY_ADVERTISING_HANDLE);` inside main.cpp.
No more unknown type errors after this fix.

It also seems to be effective as with `HciVsA3_SetRfPowerLevelEx(TX_POWER_LEVEL_MINUS_10P0_dBm);` both power consumption and also received signal strength is lower.
#### Impact of changes
closed 2021-04-28 by 0xc0170