ARMmbed/mbed-os #14591: WISUN framecounter error

mcuxmx, 2021-04-26:
### Description of defect
I built the wisun network. 
If I configure nanostack- hal.use - kvstore:true. When node restarts continuously (about every 2 minutes), it will not be able to connect to BR. The following log will appear.
[ERR][wspc]: Frame counter space exhausted.

If I configure nanostack- hal.use - kvstore:false. When EAP-TLS is completed (before RPL process) restart node. Node also cannot connect to BR.

#### Target(s) affected by this defect ?
BR: DISCO_F769NI+S2LP radio
NODE: NUCLEO_F429ZI+S2LP radio

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
any

#### How is this defect reproduced ? 
alyways

mcuxmx, 2021-04-26:
@ciarmcom I'm sorry I missed it. I used mbed cli 1.10.4

0xc0170, 2021-04-26:
@artokin Please review

mikaleppanen, 2021-05-06:
Trace originates here:

https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_pae_controller.c#L815

It means that whole frame counter range for the GTK has been used. On reboot the counter range is incremented by FRAME_COUNTER_INCREMENT which is 1000000 so for 32 bit counters exhaustion needs 4294 restarts. 

@mcuxmx Is it possible that hw could have reached this limit? 

In this case either node KV store needs to be cleared or the border router needs to be forced to do GTK update to refresh the frame counter range (either by clearing the border router KV store or by forcing the GTK update).

mcuxmx, 2021-05-08:
@mikaleppanen, Thank you for your reply.
Normally, this does not happen. But our device has a stress test: restart every three seconds.
Whether it is possible to use KV store only for Router and not for BorderRouter. Because I test that there is no problem in this configuration.

mikaleppanen, 2021-05-10:
@mcuxmx The security key counters are stored via KV store to flash before they reach threshold on FRAME_COUNTER_STORE_THRESHOLD, and the counters are incremented in restart the amount of FRAME_COUNTER_INCREMENT (and stored to flash). Depending on use case and hardware that limit can be changed. If the flash or other permanent memory system supports large amount of writes before wearing, then the increment and threshold can be set smaller. E.g. to 50000.

https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/nanostack/sal-stack-nanostack/source/6LoWPAN/ws/ws_config.h#L205

If testing only power up/down use case then the counter incrementing can be disabled fully by setting FRAME_COUNTER_INCREMENT value to zero.

mcuxmx, 2021-05-10:
@mikaleppanen 
_and the counters are incremented in restart the amount of FRAME_COUNTER_INCREMENT (and stored to flash)._

Why not store it wait for a storage interval FRAME_COUNTER_STORE_INTERVAL or after connecting. I don't think the framecounter stored at this time is what we expected.

mcuxmx, 2021-05-11:
@mikaleppanen 
 Thank you. Look forward to the future version

0xc0170, 2021-12-23:
@mikaleppanen was the feature released actually and this can be closed?

mikaleppanen, 2021-12-27:
@0xc0170 Yes, the improvement for this has been done and released. This can be closed.

closed 2021-12-29 by 0xc0170