ARMmbed/mbed-os #14596: STM CAN Slave Start Filter Bank Set Incorrectly

cindyli-13, 2021-04-26:
### Description of defect

For STM devices that make use of dual CAN instances, a "slave start filter bank" number must be set to indicate that all filter banks with a lower index are dedicated to the master CAN instance while all filter banks with a greater index are dedicated to the slave CAN instance. 

In the file `mbed-os/targets/TARGET_STM/can_api.c`, function `can_filter()`, the slave start filter bank is incorrectly set on this line: `sFilterConfig.BankNumber = 14 + handle;`. Here, `sFilterConfig.BankNumber` is the slave start filter bank and `handle` refers to the filter bank number, meaning that filters cannot be configured for the slave CAN instance since the slave start filter bank will always be greater than `handle`. 

The solution is to change the line to `sFilterConfig.BankNumber = 14;` to set the slave start filter bank to 14. This way, filters can be configured for the slave CAN instance by passing in a `handle` greater than or equal to 14.

#### Target(s) affected by this defect ?

All STM targets

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.5.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A

#### How is this defect reproduced ? 

Passing any value of `handle` to `CAN::filter()` will not configure the filter for the slave CAN instance.

0xc0170, 2021-04-27:
@ARMmbed/team-st-mcd Please review

jeromecoutant, 2021-04-30:
@0xc0170 I think you can set the st device label

@cindyli-13 It seems you have a solution to share in a pull request ! :-)
We will be happy to review it and merge!
Thx

PS: Maybe linked to #11953 ?

jeromecoutant, 2021-06-15:
@cindyli-13 maybe we can close this issue as #14612 is merged ?
Thx

closed 2021-06-15 by cindyli-13