ARMmbed/mbed-os #14598: Operator delete coud fail  - Doesn't check for unvalid pointer.

MariwanJ, 2021-04-27:
### Description of defect
If you delete any pointer using command "delete", mbed-os/platform/source/mbed_retarget.cpp" has the implementation which is clear that it doesn't check if the pointer is valid. 
Other implementations might be OK but mbed implementation is not .. As the following quesion answer the question, it must be a bug in MBED. 

**https://stackoverflow.com/questions/615355/is-there-any-reason-to-check-for-a-null-pointer-before-deleting**

Here the portion of the code : 

```
void operator delete (void *ptr)
{
    free(ptr);
}

void operator delete (void *ptr, std::size_t)
{
    free(ptr);
}

void operator delete[](void *ptr)
{
    free(ptr);
}

void operator delete[](void *ptr, std::size_t)
{
    free(ptr);
}
```

#### Target(s) affected by this defect ?
MBED-OS .. All targets as it is a base (c++) function.

#### Toolchain(s) (name and version) displaying this defect ?

GNU-ARM 
#### What version of Mbed-os are you using (tag or sha) ?

```

#define MBED_MAJOR_VERSION 6

/** MBED_MINOR_VERSION
  * Mbed OS minor version
  *
  * @note 99 is default value for development version (master branch)
  */
#define MBED_MINOR_VERSION 8

/** MBED_PATCH_VERSION
  * Mbed OS patch version
  *
  * @note 99 is default value for development version (master branch)
  */
#define MBED_PATCH_VERSION 0

```
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
eclipse, make-gnu, export from online IDE.

#### How is this defect reproduced ? 

Any time you delete a pointer, you have to check that is not NULL .. This shouldn't be necessary.
It is just normal that you SHOULD check a pointer before deleting. If I use only free command, I will check for the availability of the pointer .. but delete shouldn't need that.

0xc0170, 2021-04-27:
Do you have any test case to prove the bug (how to reproduce the problem ? 

Is this only GCC Arm or also Armclang ? 

If you look at `free_wrapper` it checks if the pointer is not null (both toolchains).

MariwanJ, 2021-04-27:
I have only GNU  compiler. I tried a smaller program at the online IDE , it didn't cause a problem. I will try to export it and test it locally. I will report you later tomorrow.

MariwanJ, 2021-04-28:
I couldn't reproduce it using another project. I might have problem in my code. But what about that peaces of code that I mentioned,  reading it tells you clearly that there is no checking for NULL pointer? Does "free" accept NULL pointer?
If you don't think it is an issue .. you can close this please. 
Thanks.

closed 2021-04-28 by 0xc0170