ARMmbed/mbed-os #14601: STM32F401xE: file name clashing

chrJost, 2021-04-28:
### Description of defect

When checking Mbed-OS out under windows, the following warning is shown:
```
warning: the following paths have collided (e.g. case-sensitive paths
on a case-insensitive filesystem) and only one from the same
colliding group is in the working tree:

  'targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE/TOOLCHAIN_GCC_ARM/STM32F401XE.ld'
  'targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE/TOOLCHAIN_GCC_ARM/stm32f401xe.ld'
```

As far as I can see, both files have the same contents, I am not sure which one is actually used.

@jeromecoutant You were the last one who made changes in that directory, do you have an idea what the correct proceeding is here?

#### Target(s) affected by this defect ?
STM32F401xE

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.10-0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli (but is independent of the actual tools)

#### How is this defect reproduced ? 
clone the mbed os repository on a windows machine or any other case-insensitive filesystem

0xc0170, 2021-04-29:
We will try to reproduce and get back to you. I haven't personally seen this when fetching mbed-os on windows but will try today to reproduce it.

0xc0170, 2021-05-17:
Closed via https://github.com/ARMmbed/mbed-os/pull/14661

closed 2021-05-17 by 0xc0170