ARMmbed/mbed-os #14606: Testing Cellular drivers/libraries

CalvoM, 2021-04-29:
### Description of defect

#### Target(s) affected by this defect ?

Currently tested on NUCLEO_F429ZI
#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.10-0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 

I am currently writing integration tests for our network components. We are using SIMCOM modem, i.e. SIM5360E, and thus we extended the Cellular Library to support SIMCOM. However, when we write the integration tests for the network functionality, the integration do not go past the __svcXX parts e.g. the __svcMutex found here. So, we sought to see how the already built Cellular libraries like Ublox are tested, so that we could find guidance, but we found none. Thus my question, does anyone have links for tests for such cellular libraries so that we could find help or even better, how to solve this svc issue we have in testing?
We are using PCB with NUCLEO-F429ZI chip and GCC_ARM toolchain.
Mbed trace is disabled.
Thanks in advance.

CalvoM, 2021-05-17:
Any progress on this?

0xc0170, 2021-05-18:
@ARMmbed/mbed-os-connectivity Please review

adbridge, 2021-05-26:
@CalvoM for future reference this is usually the kind of question that might be best asked in our forums https://forums.mbed.com/c/mbed-os/6 .