ARMmbed/mbed-os #14611: [BLE] When privacy is enabled, unable to use address type other than RANDOM

AGlass0fMilk, 2021-04-30:
### Description of defect

When privacy is enabled, there is a check in `Gap::setAdvertisingParameters` that prevents advertising an address type other than `RANDOM`

The trace message says "privacy enabled, cannot use random address", but the check itself ensures the opposite!

What is the intent of this check?

It seems like it should be entirely possible to use the other address types, `RESOLVABLE_PRIVATE_ADDRESS_PUBLIC_FALLBACK` and `RESOLVABLE_PRIVATE_ADDRESS_RANDOM_FALLBACK`...

I'm once again having pairing/bonding issues with BLE and I think it is related to an inconsistent advertised address type and the actual address type used.

#### Target(s) affected by this defect ?
BLE targets

#### Toolchain(s) (name and version) displaying this defect ?
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

#### What version of Mbed-os are you using (tag or sha) ?

 mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
Package             Version     Location
------------------- ----------- -------------------------------------------------------------------
appdirs             1.4.3
asn1ate             0.6.0
attrs               19.3.0
Automat             20.2.0
beautifulsoup4      4.6.3
ble-serial          1.2.1
bleak               0.9.1
bluepy              1.3.0
cbor                1.0.0
certifi             2019.11.28
cffi                1.14.1
chardet             3.0.4
click               7.1
cmsis-pack-manager  0.2.10
cobs                1.1.4
cogapp              3.0.0
colorama            0.3.9
constantly          15.1.0
crc16               0.1.1
crccheck            0.6
cryptography        2.9.2
cycler              0.10.0
dataclasses         0.8
distlib             0.3.1
docopt              0.6.2
ecdsa               0.15
elftools            0.1.0.dev0
ep                  0.0.1       /home/gdbeckstein/Documents/embeddedplanet/ep-app-generator
fasteners           0.15
filelock            3.0.12
Flask               1.1.2
future              0.16.0
futures             3.1.1
fuzzywuzzy          0.18.0
gitdb               4.0.5
GitPython           3.1.13
grip                4.5.2
hidapi              0.9.0.post2
hyperlink           20.0.1
icetea              1.2.4
idna                2.7
imgtool             1.7.0rc1
importlib-metadata  1.6.0
importlib-resources 5.1.2
incremental         17.5.0
inflection          0.5.1
iniconfig           1.1.1
intelhex            2.2.1
itsdangerous        1.1.0
Jinja2              2.10.3
jsonmerge           1.7.0
jsonschema          2.6.0
junit-xml           1.8
keyrings.alt        4.0.2
kiwisolver          1.2.0
lockfile            0.12.2
Logbook             1.5.3
Mako                1.1.4
manifest-tool       1.5.2
Markdown            3.3.2
MarkupSafe          1.1.1
matplotlib          3.3.0
mbed-ble-test-suite 0.0.1       /home/gdbeckstein/Documents/mbed-os-bluetooth-integration-testsuite
mbed-cli            1.10.4
mbed-cloud-sdk      2.0.8
mbed-flasher        0.10.1
mbed-greentea       1.7.4
mbed-host-tests     1.5.10
mbed-ls             1.7.12
mbed-os-tools       0.0.15
mbed-tools          7.1.2
milksnake           0.1.5
monotonic           1.5
numpy               1.19.1
packaging           20.4
path-and-address    2.0.1
pc-ble-driver-py    0.14.2
pdoc3               0.9.2
Pillow              7.2.0
pip                 20.2
pkg-resources       0.0.0
pluggy              0.13.1
prettytable         0.7.2
protobuf            3.5.2.post1
psutil              5.6.6
py                  1.9.0
pyasn1              0.2.3
pycparser           2.20
pycryptodome        3.9.8
pyelftools          0.25
Pygments            2.7.1
PyHamcrest          2.0.2
pyparsing           2.4.7
pyrsistent          0.16.0
pyserial            3.4
pytest              6.1.1
python-dateutil     2.8.1
python-dotenv       0.14.0
pyudev              0.22.0
pyusb               1.0.2
PyYAML              4.2b1
requests            2.20.1
semver              2.10.2
setuptools          46.1.3
six                 1.12.0
smmap               3.0.5
soupsieve           2.0
tabulate            0.8.9
toml                0.10.1
tqdm                4.57.0
trollius            2.1.post2
Twisted             20.3.0
txdbus              1.1.2
typing-extensions   3.7.4.3
urllib3             1.24.2
virtualenv          20.4.3
Werkzeug            1.0.1
wheel               0.34.2
wrapt               1.12.1
yattag              1.13.2
zipp                3.1.0
zope.interface      5.2.0
```

#### How is this defect reproduced ? 

Attempt to set an advertised address type other than `RANDOM` when privacy is enabled.

@pan- @paul-szczepanek-arm @noonfom

paul-szczepanek-arm, 2021-04-30:
I think the trace message is just wrong. When privacy is enabled the advertising address must be random.

paul-szczepanek-arm, 2021-04-30:
But I'm very much interested in your pairing/bonding issues. Please tell us about those.

AGlass0fMilk, 2021-05-18:
Closing, continuation of pairing/bonding issues in #14654

closed 2021-05-18 by AGlass0fMilk