ARMmbed/mbed-os #14615: standard Pinnames like SPI_MISO or I2C_SCL are missing in Mbed OS 6.10.0

JojoS62, 2021-05-03:
### Description of defect

Standard Pinnames like these were defined for many targets in Pinnames.h:

```
    I2C_SCL     = PB_8,
    I2C_SDA     = PB_9,
    SPI_MOSI    = PB_15,
    SPI_MISO    = PB_14,
    SPI_SCK     = PD_3,
    SPI_CS      = PH_6,
```

Defines have moved to hal/include/PinNameAlases.h, but renamed for Arduino FF.

These changes should be compatible, for libraries and apps that use defines like SPI_MOSI it is a nightmare to follow the os version dependent changes.

#### Target(s) affected by this defect ?

DISCO_F469NI and many more

#### Toolchain(s) (name and version) displaying this defect ?

gcc_arm

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.10.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli

#### How is this defect reproduced ? 

```
SPI spi(SPI_MISO, SPI_MOSI, SPI_SCK); // use spi with default PinNames
```

0xc0170, 2021-05-03:
@JojoS62 Is this considered breaking an applications ? The pinname pull request https://github.com/ARMmbed/mbed-os/pull/14381 should be non breaking. 
I checked also docs page https://github.com/ARMmbed/mbed-os-5-docs/pull/1417  , it does not contain any information for SPI/I2C pins change.

@MarceloSalazar  @gpsimenos Please review

JojoS62, 2021-05-03:
yes, I have compiled for target DISCO_F469NI, with mbed-os-6.9.0 it works, with 6.10.0 it complains about missing SPI_xxx.

jeromecoutant, 2021-05-04:
@MarceloSalazar Question:

Should I add
#define I2C_SDA    ARDUINO_UNO_I2C_SDA
#define I2C_SCL     ARDUINO_UNO_I2C_SCL
etc....
in each PinNames.h file

or in https://github.com/ARMmbed/mbed-os/blob/master/hal/include/hal/PinNameAliases.h
?

Thx

jeromecoutant, 2021-05-04:
I would choose to update each PinNames
as a board can propose different "connector", and it is up to target to choose the default connector behind I2C_SDA/SCL aliases ?

Better way should be to choose this default I2C/SPI aliases in some json files, that can be overwritten by application ?

gpsimenos, 2021-05-04:
> I would choose to update each PinNames
> as a board can propose different "connector", and it is up to target to choose the default connector behind I2C_SDA/SCL aliases ?
> 
> Better way should be to choose this default I2C/SPI aliases in some json files, that can be overwritten by application ?

I agree. In the meantime, I opened a PR to add aliases for I2C/SPI until those are defined in each PinNames.h file.

#14617

jeromecoutant, 2021-06-09:
@JojoS62 Should be OK now with #14617 

#ifndef SPI_MOSI
#define SPI_MOSI ARDUINO_UNO_SPI_MOSI
#endif

SPI_MOSI is defined by default to the Arduino pin, but you could still define in you board pinnames.h file

JojoS62, 2021-06-09:
yes, thanks, it works again with SPI_MOSI !

closed 2021-06-09 by JojoS62