ARMmbed/mbed-os #14616: STM32G474 freezes due to code size

pilotak, 2021-05-04:
### Description of defect
When you want to run a bigger code (> ~125k) on STM32G474, the program never runs. I made a code to make it reproduceable, please see at the bottom.

If you adjust the size of array to `60000` it runs fine, if you adjust to `62800` it prints following error, if you adjust to `70000` it freezes, no serial debug is available.

I use STM32G474RC which has 256k of ROM, code was compiled for both NUCLEO_G474RE and custom target with adjusted ROM size with the same result. On STM32F091RC there is no problem at all.

If you move the array to RAM (by removing word `const`) it runs fine. So i believe this is ROM related.

```
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : 200014DC
R1   : 0000000B
R2   : 200014C8
R3   : E9CDA818
R4   : 00000001
R5   : 20000D4C
R6   : 00000001
R7   : 00000000
R8   : 00000000
R9   : 00000000
R10  : 00000000
R11  : 00000000
R12  : 08000E41
SP   : 200027A0
LR   : 0800689B
PC   : E9CDA818
xPSR : 00000000
PSP  : 20002780
MSP  : 2001FFC0
CPUID: 410FC241
HFSR : 40000000
MMFSR: 00000001
BFSR : 00000000
UFSR : 00000000
DFSR : 00000008
AFSR : 00000000
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0xE9CDA818
Error Value: 0x200027F8
Current Thread: main Id: 0x20001530 Entry: 0x8001B35 StackSize: 0x1000 StackMem: 0x200017B8 SP: 0x200027A0
Next:
main  State: 0x2 Entry: 0x08001B35 Stack Size: 0x00001000 Mem: 0x200017B8 SP: 0x20002778
Ready:
rtx_idle  State: 0x1 Entry: 0x08001C25 Stack Size: 0x00000300 Mem: 0x200011C8 SP: 0x20001488
Wait:
rtx_timer  State: 0x83 Entry: 0x080019F1 Stack Size: 0x00000300 Mem: 0x20000EC8 SP: 0x20001160
Delay:
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NUCLEO_G474RE
-- MbedOS Error Info --
```

#### Target(s) affected by this defect ?
STM32G474

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM 9.3.1

#### What version of Mbed-os are you using (tag or sha) ?

376fda5bf59c37bfcc0615e6a902c9ac74a96e88

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.1

#### How is this defect reproduced ? 
```cpp
#include "mbed.h"

const uint8_t t[70000] = {0};

int main() {
    printf("start\n");

    for (auto i = 0; i < 70000; i++) {
        printf("%02X ", t[i]);
        fflush(stdout);
    }

    printf("\n");
    return 0;
}
```

0xc0170, 2021-05-04:
@ARMmbed/team-st-mcd Please review

AGlass0fMilk, 2021-05-05:
@pilotak `const` doesn't necessarily guarantee that code ends up in ROM or RAM.

Could you post the build output? When you use the `const` keyword does the binary size become drastically smaller?

AGlass0fMilk, 2021-05-05:
Also, you should add an infinite while loop after your code ends.

When main returns in Mbed this usually means the system has crashed and Mbed might do some funny things.

pilotak, 2021-05-05:
Interestingly buffer size is counted up to final size nor the static RAM, but with removing the `const` it will enlarge the RAM in mbed output so I suppose with `const` it's in flash.

I didn't know how else to make a fw larger in order to keep it simple to reproduce. Obviously i work on other fw and noticed the code size problem (same code is compiled for F091RC without problems) so i started a new project and i was trying to replicate it.

I added a the infinite loop, but with the same result.

```
Building project test (NUCLEO_G474RE, GCC_ARM)
Scan: test
Link: test
Elf2Bin: test
| Module           |     .text |    .data |     .bss |
|------------------|-----------|----------|----------|
| [fill]           |    82(+0) |    8(+0) |   29(+0) |
| [lib]\c.a        | 32604(+0) | 2472(+0) |   89(+0) |
| [lib]\gcc.a      |  3104(+0) |    0(+0) |    0(+0) |
| [lib]\misc       |   188(+0) |    4(+0) |   28(+0) |
| [lib]\nosys.a    |    32(+0) |    0(+0) |    0(+0) |
| main.o           |    80(+0) |    0(+0) |    0(+0) |
| mbed-os\cmsis    |  6942(+0) |  168(+0) | 6212(+0) |
| mbed-os\drivers  |   140(+0) |    0(+0) |    0(+0) |
| mbed-os\hal      |  1606(+0) |    8(+0) |  114(+0) |
| mbed-os\platform |  4612(+0) |  260(+0) |  424(+0) |
| mbed-os\targets  | 10530(+0) |    8(+0) | 1064(+0) |
| Subtotals        | 59920(+0) | 2928(+0) | 7960(+0) |
Total Static RAM memory (data + bss): 10888(+0) bytes
Total Flash memory (text + data): 62848(+0) bytes

Image: .\BUILD\NUCLEO_G474RE\GCC_ARM\test.bin
```
`mbed_app.json`
```json
{
    "target_overrides": {
      "*": {
        "platform.error-all-threads-info": true,
        "platform.error-filename-capture-enabled": true,
        "platform.fatal-error-auto-reboot-enabled": true,
        "platform.stdio-convert-newlines": true,
        "platform.stdio-baud-rate": 115200,
        "platform.callback-nontrivial": true,
        "target.printf_lib": "std"
      },
      "NUCLEO_G474RE": {
        "target.stdio_uart_tx": "PB_3",
        "target.stdio_uart_rx": "PB_4",
        "target.clock_source": "USE_PLL_HSE_XTAL"
      }
    }
  }
```

pilotak, 2021-05-07:
@AGlass0fMilk got any ideas?

AGlass0fMilk, 2021-05-08:
@pilotak I'm skeptical that the `const` array is actually increasing flash size to test this. Are you able to debug with a JLink or some other debug probe? You could pause execution after the crash and see where the processor is.

In your original, proprietary code that is above 128kB in flash, when does it freeze? Does it just never start at all?

My gut feeling is that this is some other error that _looks_ like it's crashing because of code size. It's pretty rare for a major chip vendor to make a mistake that would cause half of an MCU's flash to be unavailable.

I would check if ST has an errata related to the STM32G474. @jeromecoutant might have some insight as well.

Any other details you can provide about your original code and the scenario under which it crashes would be helpful. The contrived example may have its own issues :/

jeromecoutant, 2021-05-10:
Hi

I had a look in the reference Manual of STM32G4.
Seems that STM32G474RC is a category 3 device.
You could check with CubeProgrammer tool, but I think you are with the default configuration which is FLASH in dual bank.
So table 7 in the doc shows that:
- for a 512K device, both banks are continuous
- for smaller device (your case), banks are not continuous..... :-(

I could see 2 solutions:
- update linker scripts to match this FLASH organisation
- use the CubeProgrammer tool to uncheck the DBANK Option Byte in order to use the continuous FLASH single bank organisation

Note that in this case, some update, may be needed in https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32G4/flash_api.c

pilotak, 2021-05-10:
Thank you @jeromecoutant for taking time looking into this. Yes you are right, i use DBANK by default.
[image]

Can i find somewhere an examle for linker script update? this is a new for me as didn't have to deal in mbed so far. How can `#undef FLASH_OPTR_DBANK` when this is directly on CubeMX included in mbed?

Let's put it this way, i don't mind which solution is the right one. I just need it to work without pathing the OS during compile time (as i do it in CI). I need whole flash to be used and not just one bank. I will be using FlashIAP for FW upgrade from external SPI flash.

AGlass0fMilk, 2021-05-12:
From the STM32G4 series reference manual, category 3 devices (such as the STM32G474) have the following flash map layout:
[image]
The problem here is that on your device with 256kB of flash, the banks are not continuous as @jeromecoutant stated above.

So your flash layout is:

`0x08000000 to 0x08020000` - First bank
`0x08040000 to 0x08060000` - Second bank

Since you're using GCC, you will need to modify the linker script here:

https://github.com/ARMmbed/mbed-os/blob/0f907d7620ef98a8db059b8a7279de22adf2fc64/targets/TARGET_STM/TARGET_STM32G4/TARGET_STM32G474xE/TOOLCHAIN_GCC_ARM/stm32g474xe.ld#L38-L42

You might want to make a new target subdirectory (`TARGET_STM32G474xC`). @jeromecoutant Perhaps the linker scripts for the G4 series should be modified to accommodate the dual-bank flash for every chip? With the current arrangement, only the chips with full memory options will work with Mbed-OS...

You can find [more information about GCC linker scripts here](https://ftp.gnu.org/old-gnu/Manuals/ld-2.9.1/html_chapter/ld_3.html#SEC16).

Essentially what you'll need to do is make the existing flash memory region smaller (so it aligns with the first bank above) and then add a second flash region that covers the second bank address range. The linker will then take care of placing code into the regions and ignoring the "dead zone" in between them.

pilotak, 2021-05-12:
@AGlass0fMilk thank you for explanation i'm with you now.

Well it would be perfect if the linker script get updated for all other people dealing with the same problem. I undestard that NUCLEOs use the biggest flash available but moving to production you don't often need that much memory and will save you couple of bucks.

@jeromecoutant could you please put together a quick fix so i can carry on working on my FW? i tried to uncheck the `DBANK` and run the example code but without success. Thank you

jeromecoutant, 2021-05-12:
Problem is I don't have this MCU...

> You might want to make a new target subdirectory (TARGET_STM32G474xC)

Yes, you have to create a new TARGET_STM32G474xC directory with the expected files inside.
I have checked CubeMx tool, but it doesn't provide any example for dual bank flash, it supposes that single bank configuration is used.

pilotak, 2021-05-12:
> Problem is I don't have this MCU...

i can send you one... 👍 

> Yes, you have to create a new TARGET_STM32G474xC directory with the expected files inside.

That is no problem i did that already. How do we proceed if no example is available?

AGlass0fMilk, 2021-05-12:
> > Problem is I don't have this MCU...
> 
> 
> 
> i can send you one... 👍 
> 

Where have you found a secret stash of STM32G-series chips!? 😁

I've had to stop designing G0's into my hobby projects because of the chip shortage ☹️

> 
> > Yes, you have to create a new TARGET_STM32G474xC directory with the expected files inside.
> 
> 
> 
> That is no problem i did that already. How do we proceed if no example is available?

The required linker script modifications are really minimal. You would probably be able to read the GCC linker docs (linked above) and implement the necessary changes before the boards you send to Jerome even clear customs. It may seem intimidating but you should be able to do it.

Try it out and see if you get anywhere. If you run into issues post your modified linker script and I'll take a look at it.

pilotak, 2021-05-12:
> Where have you found a secret stash of STM32G-series chips!?

6 weeks ago our supplier discovered 150pcs of `RC` version so we took them all, `RE` is sold out everywhere. It's a nightmare making a product this year.

> Try it out and see if you get anywhere. If you run into issues post your modified linker script and I'll take a look at it.

Thank you for support, i will give it a go

pilotak, 2021-05-13:
a quick update - i did try to adjust the linker script. But to my undestanding if you have two banks you can't simply "follow" the previous one - you need to adjust the main code to and specify which data will located in second bank.
```
MEMORY
{
    FLASH1 (rx)   : ORIGIN = 0x8000000, LENGTH = 0x20000
    FLASH2 (rx)   : ORIGIN = 0x8040000, LENGTH = 0x20000
    RAM (rwx)    : ORIGIN = MBED_RAM_START + VECTORS_SIZE, LENGTH = MBED_RAM_SIZE - VECTORS_SIZE
}
SECTIONS
{
  .text :
    {
        KEEP(*(.isr_vector))
        *(.text*)

        KEEP(*(.init))
        KEEP(*(.fini))

        /* .ctors */
        *crtbegin.o(.ctors)
        *crtbegin?.o(.ctors)
        *(EXCLUDE_FILE(*crtend?.o *crtend.o) .ctors)
        *(SORT(.ctors.*))
        *(.ctors)

        /* .dtors */
        *crtbegin.o(.dtors)
        *crtbegin?.o(.dtors)
        *(EXCLUDE_FILE(*crtend?.o *crtend.o) .dtors)
        *(SORT(.dtors.*))
        *(.dtors)

        *(.rodata*)

        KEEP(*(.eh_frame*))
    } > FLASH1

.Flash2 :
    {
        *(Flash2 Flash2.*)
		*(.gnu.linkonce.r.*)
    } > FLASH2
}
```
`main.cpp`
```cpp
__attribute__((section(".Flash2"))) const uint8_t t[70000] = {0};
```
Problem with this code is that generated bin file is too big to flash (probably i missed somehing)

Anyway i tried the second approach and disabled `DBANK` attribute. I know i wrote that it doesn't work - well it DOES! the reason why it didn't work before is that you need to power cycle the processor not just do a pin reset, or maybe it was due to a fully charged backup capacitor on VBAT pin.

jeromecoutant, 2021-05-17:
> Problem with this code is that generated bin file is too big to flash (probably i missed somehing)

Maybe there is some padding between 0x8020000 and 0x8040000 ?

Maybe HEX format could be tried ?
ex: https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L3118

pilotak, 2021-05-19:
unfortunatelly output is still too big, HEX has 381kB, BIN 325kB

AGlass0fMilk, 2021-05-19:
> unfortunatelly output is still too big, HEX has 381kB, BIN 325kB

A hex file (typically using the [Intel Hex](https://en.m.wikipedia.org/wiki/Intel_HEX) format) is a text file containing binary information.

The advantage of a hex file is that, unlike with binary files, you can have different memory regions and discontinuous binary information.

Essentially, the hex file will say "put the first 128kB at address 0x80020000, then skip from 0x80040000 to 0x80060000, then put the next 128kB there"

(I just made up the addresses, I can't remember what they are in your case and I'm typing on the GH mobile app)

Hex files are often much larger than the binary they are specifying, but that's because they contain extra information.

How did you get the hex file output from the build?

AGlass0fMilk, 2021-05-19:
Unfortunately, I can't find any support for discontinuous memory regions like this in the GCC linker...

What you might try is specifying a lot of library code you use, such as mbed-os, goes into the first memory region, and the rest of your code goes into the second memory region.

See this: https://stackoverflow.com/questions/29437877/splitting-embedded-program-in-multiple-parts-in-memory

jeromecoutant, 2021-05-19:
I am afraid that dual bank feature can not be supported for STM32G474RC ...
It can be useful only for FOTA for ex
To get the full flash, disable the dual bank feature...

pilotak, 2021-05-19:
> How did you get the hex file output from the build?

exactly like that https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L3118

Thank you @jeromecoutant @AGlass0fMilk for support, the result is to disable `DBANK` bit in order to use whole flash

closed 2021-05-19 by pilotak