ARMmbed/mbed-os #14621: [BLE] Authorize read request ignores nullptr data param

AGlass0fMilk, 2021-05-05:
### Description of defect

When using the BLE stack feature that allows you to register read authorization callbacks on a GattCharacteristic, the documentation comment states the following:

```
     * @note If the read request is approved and params->data remains nullptr, then
     * the current characteristic value is used in the read response payload.
```
[Link](https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/FEATURE_BLE/include/ble/gatt/GattCharacteristic.h#L1682-L1683)

In my handler, I explicitly set the `params->data` output field to `nullptr` so the current characteristic value is used rather than whatever my read authorization handler outputs. This is not having any effect. Characteristics that have read authorization callbacks simply read back 0 bytes.

It seems from the GattServer implementation code that this documented condition is not respected:

https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/FEATURE_BLE/source/cordio/source/GattServerImpl.cpp#L1149-L1150

#### Target(s) affected by this defect ?

All BLE targets

#### Toolchain(s) (name and version) displaying this defect ?

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

```
Package             Version     Location
------------------- ----------- -------------------------------------------------------------------
appdirs             1.4.3
asn1ate             0.6.0
attrs               19.3.0
Automat             20.2.0
beautifulsoup4      4.6.3
ble-serial          1.2.1
bleak               0.9.1
bluepy              1.3.0
cbor                1.0.0
certifi             2019.11.28
cffi                1.14.1
chardet             3.0.4
click               7.1
cmsis-pack-manager  0.2.10
cobs                1.1.4
cogapp              3.0.0
colorama            0.3.9
constantly          15.1.0
crc16               0.1.1
crccheck            0.6
cryptography        2.9.2
cycler              0.10.0
dataclasses         0.8
distlib             0.3.1
docopt              0.6.2
ecdsa               0.15
elftools            0.1.0.dev0
ep                  0.0.1       /home/gdbeckstein/Documents/embeddedplanet/ep-app-generator
fasteners           0.15
filelock            3.0.12
Flask               1.1.2
future              0.16.0
futures             3.1.1
fuzzywuzzy          0.18.0
gitdb               4.0.5
GitPython           3.1.13
grip                4.5.2
hidapi              0.9.0.post2
hyperlink           20.0.1
icetea              1.2.4
idna                2.7
imgtool             1.7.0rc1
importlib-metadata  1.6.0
importlib-resources 5.1.2
incremental         17.5.0
inflection          0.5.1
iniconfig           1.1.1
intelhex            2.2.1
itsdangerous        1.1.0
Jinja2              2.10.3
jsonmerge           1.7.0
jsonschema          2.6.0
junit-xml           1.8
keyrings.alt        4.0.2
kiwisolver          1.2.0
lockfile            0.12.2
Logbook             1.5.3
Mako                1.1.4
manifest-tool       1.5.2
Markdown            3.3.2
MarkupSafe          1.1.1
matplotlib          3.3.0
mbed-ble-test-suite 0.0.1       /home/gdbeckstein/Documents/mbed-os-bluetooth-integration-testsuite
mbed-cli            1.10.4
mbed-cloud-sdk      2.0.8
mbed-flasher        0.10.1
mbed-greentea       1.7.4
mbed-host-tests     1.5.10
mbed-ls             1.7.12
mbed-os-tools       0.0.15
mbed-tools          7.1.2
milksnake           0.1.5
monotonic           1.5
numpy               1.19.1
packaging           20.4
path-and-address    2.0.1
pc-ble-driver-py    0.14.2
pdoc3               0.9.2
Pillow              7.2.0
pip                 20.2
pkg-resources       0.0.0
pluggy              0.13.1
prettytable         0.7.2
protobuf            3.5.2.post1
psutil              5.6.6
py                  1.9.0
pyasn1              0.2.3
pycparser           2.20
pycryptodome        3.9.8
pyelftools          0.25
Pygments            2.7.1
PyHamcrest          2.0.2
pyparsing           2.4.7
pyrsistent          0.16.0
pyserial            3.4
pytest              6.1.1
python-dateutil     2.8.1
python-dotenv       0.14.0
pyudev              0.22.0
pyusb               1.0.2
PyYAML              4.2b1
requests            2.20.1
semver              2.10.2
setuptools          46.1.3
six                 1.12.0
smmap               3.0.5
soupsieve           2.0
tabulate            0.8.9
toml                0.10.1
tqdm                4.57.0
trollius            2.1.post2
Twisted             20.3.0
txdbus              1.1.2
typing-extensions   3.7.4.3
urllib3             1.24.2
virtualenv          20.4.3
Werkzeug            1.0.1
wheel               0.34.2
wrapt               1.12.1
yattag              1.13.2
zipp                3.1.0
zope.interface      5.2.0

```

#### How is this defect reproduced ? 

Attempt to return a `nullptr` as the `params->data` field in a read authorization callback so that the stack uses the current value of characteristic. Instead, the GattServer implementation will reply to the GattClient with 0 bytes.

AGlass0fMilk, 2021-05-05:
@pan- @paul-szczepanek-arm @noonfom

paul-szczepanek-arm, 2021-05-05:
It's even worse than that, the callback overwrites the attribute in the server. Thanks for spotting this.

noonfom, 2021-05-05:
Does it not makes sense to overwrite the attribute? Otherwise, what would be the point of the data and length fields in [`GattReadAuthCallbackParams`](https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/FEATURE_BLE/include/ble/gatt/GattCallbackParamTypes.h#L355) ? Perhaps, we can simply check [`read_auth_params.data`](https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/FEATURE_BLE/source/cordio/source/GattServerImpl.cpp#L1129) and only update the attribute if it's not `NULL` ?

AGlass0fMilk, 2021-05-05:
> Does it not make sense to overwrite the attribute? Otherwise, what would be the point of the data and length fields in [`GattReadAuthCallbackParams`](https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/FEATURE_BLE/include/ble/gatt/GattCallbackParamTypes.h#L355) ? Perhaps, we can simply check [`read_auth_params.data`](https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/FEATURE_BLE/source/cordio/source/GattServerImpl.cpp#L1129) and only update the attribute if it's not `NULL`.

In my case, I’m running everything in bare metal in a single thread. The characteristic data comes from another module over a UART. I implemented the UART protocol to be non-blocking because I’m in bare metal. So the problem for me was that I wanted to use the cached data in the characteristic. Instead, I had to implement a workaround to block in the read authorization callback until the UART transaction was complete and then return the data using`params`.

This actually ended up working better for my application. In some cases though, the user may want to just authorize a read without changing the data in the characteristic.

The doc comment says that you can leave the data pointer null to use the current characteristic value. The GattServer implementation should be patched to reflect this.

It would also be good to update the BLE examples to call out this functionality (returning nullptr vs providing data in the callback). I had to read the documentation closely when it wasn’t working as expected but others may not dig that deep and miss this nuance of the API.

paul-szczepanek-arm, 2021-05-05:
The current version is wrong. I am currently fixing this. I'll invite you to the PR review in a moment.

noonfom, 2021-05-05:
@AGlass0fMilk Thanks for the explanation. That makes a lot of a sense and I completely agree that the Gatt Server doesn't respect the [condition](https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/FEATURE_BLE/include/ble/gatt/GattCharacteristic.h#L1682-L1683). I should have been clearer in my post that I was responding to Paul's [comment](https://github.com/ARMmbed/mbed-os/issues/14621#issuecomment-832555083) about the callback overwriting the attribute. It makes sense to overwrite the attribute, but you should be able **not** to by writing `NULL` to `params->data`. 

> The doc comment says that you can leave the data pointer null to use the current characteristic value. The GattServer implementation should be patched to reflect this.
> 

I agree. The check could be made in [atts_read_cb](https://github.com/ARMmbed/mbed-os/blob/376fda5bf59c37bfcc0615e6a902c9ac74a96e88/connectivity/FEATURE_BLE/source/cordio/source/GattServerImpl.cpp#L1119) before the attribute is written.

paul-szczepanek-arm, 2021-05-05:
https://github.com/ARMmbed/mbed-os/pull/14625

paul-szczepanek-arm, 2021-05-10:
fix merged

closed 2021-05-10 by paul-szczepanek-arm