ARMmbed/mbed-os #14622: platform.stack-dump-enabled makes STM targets not build

rotu, 2021-05-05:
### Description of defect

Setting the target overrides with ` "platform.stack-dump-enabled": 1` causes mbed-os to fail to build with the following error:

```
uint32_t msp_size = MAX(0, (int)INITIAL_SP - (int)msp_sp);
^
.\mbed-os\platform\source\mbed_error.c:521:41: error: use of undeclared identifier 'INITIAL_SP'
```

Maybe instead of defining `INITIAL_SP` for each target, it would be more correct to use the compiler's top-of-stack, e.g.:
ARMCC: `Image$$ARM_LIB_STACK$$ZI$$Base`
GCC: `__StackTop`

#### Target(s) affected by this defect ?

All STM targets.

#### Toolchain(s) (name and version) displaying this defect ?

ARMC6

#### What version of Mbed-os are you using (tag or sha) ?

376fda5bf59c37bfcc0615e6a902c9ac74a96e88

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

MBed Studio 1.4.0

#### How is this defect reproduced ? 

see above description

0xc0170, 2021-05-05:
Thanks @jeromecoutant for connecting the older issue. Let's take it to #14432 . I'll close this as duplicate.

closed 2021-05-05 by 0xc0170

jeromecoutant, 2021-05-05:
@rotu check #14624