ARMmbed/mbed-os #14633: ARMClang assembly files: use gcc versions (gnu syntax) instead of arm files

0xc0170, 2021-05-07:
### Description of defect

We hit an issue with armasm (invoked via armclang for .S files due to using old arm syntax) and v8m. CMSIS uses already Gcc Arm files (gnu syntax) and it works. We should do the same to avoid having similar issues in the future.

#### Target(s) affected by this defect ?

v8m targets

#### Toolchain(s) (name and version) displaying this defect ?

ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

not relevant

#### How is this defect reproduced ? 

Build cortex-m55 mcu (need a custom target at  the moment)

Patater, 2021-05-13:
https://github.com/ARM-software/CMSIS_5/issues/984#issuecomment-673263904 recommends we use C-based startup instead of assembly for ARMC6 targets. So, the CMSIS-supported solution seems to be to use the C-based startup for ARMC6, rather than to use the GCC startup asm files with the ARMC6 toolchain, as @kristofer-jonsson-arm's workaround for M55 does.

Patater, 2021-05-14:
It seems we don't use asm startup. We use legacy assembler syntax for `irq_armv8mml.S` (from CMSIS) and `except.S` (our own file). We likely need a new-syntax Armclang `irq_armv8mml.S`.

PR #14667 (2021-05-27): Update GCC except.S to support ARMC6 and use GCC assembly with ARM toolchain in CMake

### Summary of changes 

Update GCC except.S to support ARM compiler, and use this source in CMake.
Update CMake to use GCC CMSIS assembly with both GCC_ARM and ARMC6.

When legacy ASM is compiled with ARM Compiler 6, `armasm` is used, which has
no support for Cortex-M55. GNU syntax is used by `armclang` assembler, so the GCC
version of except.S is updated to support the ARM toolchain, and this is used by
CMake for both ARM and GCC_ARM. The CMSIS GCC assembly is also used now
with ARMC6.

Fixes #14633

####  [... cut]
closed 2021-05-27 by Patater