ARMmbed/mbed-os #14648: WiSUN ping echo delay varies from 90ms to 400ms periodically

mcuxmx, 2021-05-08:
### Description of defect
WiSUN ping echo delay varies from 90ms to 400ms periodically. When BR or Roter sends data (such as PC, DIO, etc.) during communication, the delay will reach 3s.
If there are multiple hops, sometimes the data access delay will be longer. I have tested a 3-hop network that often has a communication delay of more than 3s. 

Ping a 1-hop node log as follows:

PING 2001::200:cd2c:107:3466(2001::200:cd2c:107:3466) 56 data bytes
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=1 ttl=63 time=99.9 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=2 ttl=63 time=123 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=3 ttl=63 time=130 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=4 ttl=63 time=160 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=5 ttl=63 time=180 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=6 ttl=63 time=197 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=7 ttl=63 time=219 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=8 ttl=63 time=242 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=9 ttl=63 time=2321 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=10 ttl=63 time=1362 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=11 ttl=63 time=523 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=12 ttl=63 time=280 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=13 ttl=63 time=302 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=14 ttl=63 time=315 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=15 ttl=63 time=347 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=16 ttl=63 time=354 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=17 ttl=63 time=364 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=18 ttl=63 time=390 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=19 ttl=63 time=413 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=20 ttl=63 time=439 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=21 ttl=63 time=441 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=22 ttl=63 time=482 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=23 ttl=63 time=491 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=24 ttl=63 time=516 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=25 ttl=63 time=526 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=26 ttl=63 time=553 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=27 ttl=63 time=568 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=28 ttl=63 time=93.3 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=29 ttl=63 time=89.7 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=30 ttl=63 time=122 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=31 ttl=63 time=123 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=32 ttl=63 time=142 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=33 ttl=63 time=174 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=34 ttl=63 time=177 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=35 ttl=63 time=202 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=36 ttl=63 time=221 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=37 ttl=63 time=248 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=38 ttl=63 time=267 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=39 ttl=63 time=275 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=40 ttl=63 time=297 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=41 ttl=63 time=319 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=42 ttl=63 time=91.1 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=43 ttl=63 time=100 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=44 ttl=63 time=135 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=45 ttl=63 time=146 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=46 ttl=63 time=165 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=47 ttl=63 time=183 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=48 ttl=63 time=2505 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=49 ttl=63 time=1998 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=50 ttl=63 time=1025 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=51 ttl=63 time=214 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=52 ttl=63 time=221 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=53 ttl=63 time=248 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=54 ttl=63 time=278 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=55 ttl=63 time=276 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=56 ttl=63 time=2653 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=57 ttl=63 time=2066 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=58 ttl=63 time=1093 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=59 ttl=63 time=314 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=60 ttl=63 time=91.5 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=61 ttl=63 time=92.6 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=62 ttl=63 time=106 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=63 ttl=63 time=125 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=64 ttl=63 time=144 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=65 ttl=63 time=163 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=66 ttl=63 time=179 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=67 ttl=63 time=3270 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=68 ttl=63 time=2320 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=69 ttl=63 time=1476 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=70 ttl=63 time=505 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=71 ttl=63 time=225 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=72 ttl=63 time=251 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=73 ttl=63 time=278 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=74 ttl=63 time=279 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=75 ttl=63 time=318 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=76 ttl=63 time=333 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=77 ttl=63 time=345 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=78 ttl=63 time=368 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=79 ttl=63 time=380 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=80 ttl=63 time=391 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=81 ttl=63 time=412 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=82 ttl=63 time=450 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=83 ttl=63 time=459 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=84 ttl=63 time=474 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=85 ttl=63 time=511 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=86 ttl=63 time=527 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=87 ttl=63 time=539 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=88 ttl=63 time=575 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=89 ttl=63 time=94.3 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=90 ttl=63 time=90.2 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=91 ttl=63 time=114 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=92 ttl=63 time=120 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=93 ttl=63 time=144 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=94 ttl=63 time=172 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=95 ttl=63 time=177 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=96 ttl=63 time=203 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=97 ttl=63 time=217 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=98 ttl=63 time=243 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=99 ttl=63 time=273 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=100 ttl=63 time=281 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=101 ttl=63 time=295 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=102 ttl=63 time=330 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=103 ttl=63 time=91.4 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=104 ttl=63 time=103 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=105 ttl=63 time=109 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=106 ttl=63 time=135 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=107 ttl=63 time=146 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=108 ttl=63 time=185 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=109 ttl=63 time=190 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=110 ttl=63 time=214 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=111 ttl=63 time=219 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=112 ttl=63 time=247 ms
64 bytes from 2001::200:cd2c:107:3466: icmp_seq=113 ttl=63 time=261 ms

#### Target(s) affected by this defect ?
BR: DISCO_F769NI+S2LP radio
NODE: NUCLEO_F429ZI+S2LP radio

#### Toolchain(s) (name and version) displaying this defect ?
any

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5.15.4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
any

#### How is this defect reproduced ? 
alyways

mikter, 2021-05-19:
There are delays in unicast traffic that will cause these types of delays for messaging.
* Sending ASYNC messages. This is a slow process as it can take even 1000+ms to go through all channels, this causes then delays as MAC level retries are slower
* Broadcast interval/dwell, by default there is 250ms broadcast time in 1050ms interval causing periodic delays
* RX/TX slot. to improve multi hop performance and reduce collisions devices have a preferred time when they send or receive. if ping comes during wrong slot it is delayed a bit perhaps 20ms-50ms

So it is quite natural to have fluctuations on individual ping packet latencies. If you want to measure performance using latency you should check Min / Max values and accept that Max can be surprisingly high and the minimum is the performance that can be achieved if there is nothing interfering the transfers

mikter, 2021-05-25:
Closing this issue as variation in ping delays is normal in the Wi-SUN network.
Please file a new issue if there is something we didn't notice from the attached log.

closed 2021-05-25 by mikter