ARMmbed/mbed-os #14650: Multiple BLE Advertising Sets do not start as described in the documentation

tjpetz, 2021-05-09:
### Description of defect

BLE Multiple Advertising Sets do not work as described in the documentation and example code

The documentation and example code indicate that to use multiple advertising sets you simply create an advertising set, set the payload of the set, and then start advertising it.  Then to create additional sets you follow the same steps with a new advertising set.

However, when you create the advertising sets and start advertising them only the first set is actually advertised.  No error codes are returned suggesting there are any problems.  However, the onAdvertisingStart() callback is called only for the first advertising set. Using a BLE scan tool confirms only the first advertising set is actually advertising.

If instead you create an advertising set and allow it to start advertising, you can then create a second set and have it start advertising.

See  https://github.com/tjpetz/Coded-Advertiser/blob/main/main.cpp for a working example.  At line 145 the legacy advertising set is created and started.  Then on line 149 after a 250ms delay the second advertising set is created and started.  (Note, I used a delay to git it enough time to start rather than creating a more complex signaling mechanism based on the onAdvertisingStart() callback).

#### Target(s) affected by this defect ?

nRF52840_DK
ARDUINO_NANO33BLE

#### Toolchain(s) (name and version) displaying this defect ?

ARM

#### What version of Mbed-os are you using (tag or sha) ?

6.10

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio 1.4.1

#### How is this defect reproduced ? 

Create and advertising set and start advertising it, then before the first set starts advertising create a second set and attempt to start advertising it.

paul-szczepanek-arm, 2021-05-13:
Thank you for reporting this. I'll have to serialise commands inside as cordio can't handle sumultanous advertising commands.

paul-szczepanek-arm, 2021-05-18:
https://github.com/ARMmbed/mbed-os/pull/14672

paul-szczepanek-arm, 2021-06-10:
Fixing this uncovered further issues. Once https://github.com/ARMmbed/mbed-os/pull/14747 is merged this should be fully fixed.

tjpetz, 2021-06-10:
Paul, thanks for the update and working to fix this (and other) issues.

paul-szczepanek-arm, 2021-06-18:
The fixes have now landed on master. Can you please check it works for you?

tjpetz, 2021-06-19:
Thanks!  The changes resolve the problem.  I've tested with 2 advertising sets and they start as advertised one after the other without the need to queue the second set to start it.

closed 2021-06-20 by paul-szczepanek-arm