ARMmbed/mbed-os #14660: [storage] kvstore_global_api.h: No such file or directory, when ble-security-database-kvstore is true

16L-YT, 2021-05-15:
### Description of defect

When I enable ble-security-database-kvstore from mbed_app.json like this: 
```
    "target_overrides": {
        "*": {
            "ble.ble-security-database-kvstore" : true
        }
```
The compilation get fail because of missing kvstore_global_api.h

```
Compile [ 68.8%]: KVStoreSecurityDb.cpp
[Fatal Error] KVStoreSecurityDb.h@24,10: kvstore_global_api.h: No such file or directory
[ERROR] In file included from .\mbed-os\connectivity\FEATURE_BLE\source\generic\KVStoreSecurityDb.cpp:21:
.\mbed-os\connectivity\FEATURE_BLE\source\generic\KVStoreSecurityDb.h:24:10: fatal error: kvstore_global_api.h: No such file or directory
24 | #include "kvstore_global_api.h"
```

Although, from the CMakeLists.txt in storage directory it seems the kvstore_global_api included correctly. 

The only other change I did is that I added `"storage_type": "TDB_INTERNAL"` from mbed-os\storage\kvstore\kv_config\global\mbed_lib.json and set storage_tdb_internal.internal_size and storage_tdb_internal.internal_base_address from storage\kvstore\kv_config\tdb_internal\mbed_lib.json

#### Target(s) affected by this defect ?
Issue does not seem to be target specific. I tried with Ambiq and Nordic targets.

#### Toolchain(s) (name and version) displaying this defect ?
gcc version 10.2.1 20201103 (release) (xPack GNU Arm Embedded GCC, 64-bit)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.4.1

#### How is this defect reproduced ? 

Compiling BLE_SecurityAndPrivacy with "ble.ble-security-database-kvstore" : true

16L-YT, 2021-05-16:
I managed to solve it by the following changes. Please check, if they are reasonable then I will make a PR. 

1- Adding the following to  mbed-os\connectivity\FEATURE_BLE\source\generic\CMakeLists.txt

```
target_link_libraries(mbed-ble
    INTERFACE
        mbed-storage-kvstore
        mbed-storage-tdbstore
        mbed-storage-kv-global-api
)
```

2- After doing the change in 1, I had a compilation error about not having a definition for `KVStoreSecurityDb::entry_t *KVStoreSecurityDb::as_entry(entry_handle_t db_handle)` so I added ` entry_t *as_entry(entry_handle_t db_handle);` to  mbed-os\connectivity\FEATURE_BLE\source\generic\KVStoreSecurityDb.h .

0xc0170, 2021-05-17:
Just to make sure, you are using mbed cli2 ? I would like to verify that and compare if we have missed the library in CMakeLists or there is more (your 2nd point in the recent comments indicates also header file is missing definition and type error).

16L-YT, 2021-05-17:
When I opened the issue initially, I was using the mbed studio, but when I solved (the modification) it I used CLI2.

noonfom, 2021-05-17:
@16L-YT I'm not sure about change (1). Perhaps, you could add `mbed-storage-kvstore` in the top-level CMakeLists.txt:

```cmake
target_link_libraries(${APP_TARGET}
        PRIVATE
        mbed-os
        mbed-events
        mbed-ble
        mbed-ble-utils
        mbed-storage-kvstore
)
```

The `as_entry` declaration is indeed missing so change (2) is required.

0xc0170, 2021-05-18:
@16L-YT  @noonfom Can you send PR with a fix?

16L-YT, 2021-05-19:
@0xc0170 I will push a PR with my proposed changes this weekend. Unless if @noonfom is going to push a PR.

16L-YT, 2021-05-22:
@0xc0170 Here: #14693

16L-YT, 2021-06-27:
Merged and fixed

closed 2021-06-27 by 16L-YT