ARMmbed/mbed-os #14666: STM32L152RE custom target pin generation failed under mbed-os-6.10.0

wiots-ff, 2021-05-17:
### Description of defect

Using the method described at https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM works for mbed-os-6.9 but not for 6.10.

#### Target(s) affected by this defect ?

STM32L152RETx custom target

#### Toolchain(s) (name and version) displaying this defect ?

None

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.9.0
vs.
mbed-os-6.10.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

STM32_gen_PeripheralPins.py 1.19
vs.
STM32_gen_PeripheralPins.py 1.20

#### How is this defect reproduced ? 

Python 3.9.5 for Windows 64 Bit

```
C:\mbedtest\mbed-os-6.9-example-blinky>python .\mbed-os\targets\TARGET_STM\tools\STM32_gen_PeripheralPins.py -m "STM32L152RETx.xml"

Script version 1.19

Checking STM32_open_pin_data repo...
*** git clone done

STM32_open_pin_data DB version STM32CubeMX-DB.6.0.20

 * Output directory: C:\mbedtest\mbed-os-6.9-example-blinky\targets_custom\TARGET_STM\TARGET_STM32L1\TARGET_STM32L152xE\TARGET_STM32L152RET
 * Generating PeripheralPins.c and PinNames.h with 'STM32_open_pin_data\mcu\STM32L152RETx.xml'
 * GPIO file: STM32_open_pin_data\mcu\IP\GPIO-STM32L162xE_gpio_v1_0_Modes.xml
 * I/O pins found: 65 connected: 0
```

```
C:\mbedtest\mbed-os-6.10-example-blinky>python .\mbed-os\targets\TARGET_STM\tools\STM32_gen_PeripheralPins.py -m "STM32L152RETx.xml"

Script version 1.20

Checking STM32_open_pin_data repo...
*** git clone https://github.com/STMicroelectronics/STM32_open_pin_data.git ***
*** git clone done

STM32_open_pin_data DB version STM32CubeMX-DB.6.0.20

 * Output directory: C:\mbedtest\mbed-os-6.10-example-blinky\targets_custom\TARGET_STM\TARGET_STM32L1\TARGET_STM32L152xE\TARGET_STM32L152RET
 * Generating PeripheralPins.c and PinNames.h with 'STM32_open_pin_data\mcu\STM32L152RETx.xml'
 * GPIO file: STM32_open_pin_data\mcu\IP\GPIO-STM32L162xE_gpio_v1_0_Modes.xml
Traceback (most recent call last):
  File "C:\mbedtest\mbed-os-6.10-example-blinky\mbed-os\targets\TARGET_STM\tools\STM32_gen_PeripheralPins.py", line 1891, in <module>
    print_header()
  File "C:\mbedtest\mbed-os-6.10-example-blinky\mbed-os\targets\TARGET_STM\tools\STM32_gen_PeripheralPins.py", line 464, in print_header
    """ % (date_year, os.path.basename(input_file_name), TargetName))
NameError: name 'TargetName' is not defined
```

jeromecoutant, 2021-05-17:
Fixed by #14594

wiots-ff, 2021-05-17:
Didn't see it in the search, thanks @jeromecoutant .

closed 2021-05-17 by wiots-ff