ARMmbed/mbed-os #14676: unmatched bracket in stm32F4 i2c_device.h

rotu, 2021-05-18:
### Description of defect
`i2c_compile.h` has an unclosed bracket:

https://github.com/ARMmbed/mbed-os/blob/14e5d307bb6cdccb554b591ab2602d8d47e0b2d0/targets/TARGET_STM/TARGET_STM32F4/i2c_device.h#L21-L23

This is not a functional aspect since the file is only included from `.c` files, not `.cpp` files.

#### Target(s) affected by this defect ?

STM32F4x

#### Toolchain(s) (name and version) displaying this defect ?

N/A

#### What version of Mbed-os are you using (tag or sha) ?

338e37abee5b0a55aa7d2a173b66ec4e767d02cd

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio 1.4.1, ARMC6

#### How is this defect reproduced ? 

Code inspection

rotu, 2021-05-18:
This same problem exists in STM32F0, STM32F1, STM32F3, and others

0xc0170, 2021-05-19:
@ARMmbed/team-st-mcd Please review

jeromecoutant, 2021-05-19:
Good catch!
@rotu do you plan to correct all the missing } you have found ?

rotu, 2021-05-19:
I am but I was not planning on adding static analysis to the codebase. This is pretty low-hanging fruit for such a tool.

rotu, 2021-05-19:
I couldn't get `cppcheck` nor `clang-check` working to check this.

Since I can't satisfy the "Test results" requirement of pull requests, I'm not planning on fixing this.

rotu, 2021-05-19:
Maybe if I knew how to build the mbed-stm CMAKE target, I could force it to treat .c files as .cpp, and verify that I got them all that way.

0xc0170, 2021-05-20:
You can use `set_source_files_properties(filename.c PROPERTIES LANGUAGE CXX )` for changed files so they would be compiled with c++ compiler instead.

If it is just i2c for multiple families, change i2c_api.c to be compiled as c++ (few families).

jeromecoutant, 2021-05-20:
I made a quick parsing script :-)
See #14686

PR #14686 (2021-06-02): STM32: correct __cplusplus brackets
### Summary of changes 

Fix #14676

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2021-06-02 by 0xc0170