ARMmbed/mbed-os #14694: Issues when compiling and executing ARIA for hardware accelerated crypto

blanca-microchip, 2021-05-24:
### Description of defect

1) Compilation error in aria.h due to wrong order of lines.

Current lines:

```
#if !defined(MBEDTLS_ARIA_ALT)
// Regular implementation
//

#ifdef __cplusplus
extern "C" {
#endif

```
Correct lines (like in aes.h):

```
#ifdef __cplusplus
extern "C" {
#endif

#if !defined(MBEDTLS_ARIA_ALT)
// Regular implementation
//

```
2) The initialization function is missing in the selftest.

```
int mbedtls_aria_self_test( int verbose )
{
    int i;
    uint8_t blk[MBEDTLS_ARIA_BLOCKSIZE];
    mbedtls_aria_context ctx;

#if (defined(MBEDTLS_CIPHER_MODE_CFB) || defined(MBEDTLS_CIPHER_MODE_CTR))
    size_t j;
#endif

#if (defined(MBEDTLS_CIPHER_MODE_CBC) || \
     defined(MBEDTLS_CIPHER_MODE_CFB) || \
     defined(MBEDTLS_CIPHER_MODE_CTR))
    uint8_t buf[48], iv[MBEDTLS_ARIA_BLOCKSIZE];
#endif

     mbedtls_aria_init( &ctx );  <--------------------------------- ADD

    /*
     * Test set 1
     */

```
#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.8.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli 1.10.4
IAR 8.32

#### How is this defect reproduced ? 

Enable ARIA with MBEDTLS_ARIA_C and compile.
Enable MBEDTLS_SELF_TEST and test.

adbridge, 2021-05-24:
@ARMmbed/mbed-tls-developers could you please take a look at this issue?

tom-daubney-arm, 2021-05-24:
Thanks @adbridge.

@blanca-microchip I am trying to reproduce. Could you please detail your build errors for me?
Thanks,
Tom

tom-daubney-arm, 2021-05-24:
Also @blanca-microchip can you please tell me what you're compiling with?

tom-daubney-arm, 2021-05-24:
I dug into mbed-os-6.8.0 and it appears to build MbedTLS v2.24. I have tried building the v2.24 with the config options you have described and I am unable the reproduce any build errors you have have had.

I built with GCC v7.5.0 and Clang v.6.0.0. Neither produced any errors or issues with the tests. I will wait for you, @blanca-microchip , to provide some more details about your development environment etc. Thanks.

gilles-peskine-arm, 2021-05-25:
@tom-daubney-arm The build error would only show up with `MBEDTLS_ARIA_ALT` defined, which requires an `aria_alt.h` header defining `mbedtls_aria_context` and an independent implementation of the `mbedtls_aria_xxx` functions.

The initialization error in the self-test is technically a bug even without `MBEDTLS_ARIA_ALT` but we didn't notice because with the built-in implementation, the setkey functions overwrite the whole context so the initialization isn't actually necessary.

Fix incoming: https://github.com/ARMmbed/mbedtls/pull/4553

tom-daubney-arm, 2021-05-25:
Thanks @gilles-peskine-arm !

gilles-peskine-arm, 2021-05-25:
The fix has been merged in Mbed TLS. You should be able to fast-track it into Mbed OS by cherry-picking from https://github.com/ARMmbed/mbedtls/pull/4557.

closed 2021-12-23 by 0xc0170