ARMmbed/mbed-os #14695: Murata Type ABZ support discontinued?

factoryal, 2021-05-24:
### Description of defect
Hi.
I'm trying to run my program on Murata Type ABZ (CMWX1ZZABZ-078) which is on my custom board.
I already tested on DISCO-L072CZ-LRWAN1(CMWX-1ZZABZ-091).

The compilation not proceed with these messages.
```
argument -m/--mcu: MTB_MURATA_ABZ is not a supported MCU. Supported MCUs are:
ADV_WISE_1510,          ADV_WISE_1570,          ARCH_MAX,
ARCH_PRO,               ARDUINO_NANO33BLE,      ARM_CM3DS_MPS2,
ARM_MPS2_M0,            ARM_MPS2_M0P,           ARM_MPS2_M3,
ARM_MPS2_M4,            ARM_MPS2_M7,            ARM_MUSCA_B1,
ARM_MUSCA_S1,           B_L4S5I_IOT01A,         CY8CKIT064B0S2_4343W,
CY8CKIT_062S2_43012,    CY8CKIT_062_BLE,        CY8CKIT_062_WIFI_BT,
CY8CPROTO_062S3_4343W,  CY8CPROTO_062_4343W,    CYSBSYSKIT_01,
CYTFM_064B0S2_4343W,    CYW9P62S1_43012EVB_01,  CYW9P62S1_43438EVB_01,
DISCO_F413ZH,           DISCO_F429ZI,           DISCO_F469NI,
DISCO_F746NG,           DISCO_F769NI,           DISCO_H747I,
DISCO_H747I_CM4,        DISCO_H747I_CM7,        DISCO_L072CZ_LRWAN1,
DISCO_L475VG_IOT01A,    DISCO_L476VG,           DISCO_L496AG,
DISCO_L4R9I,            DISCO_L562QE,           EFM32GG11_STK3701,
EFM32GG_STK3700,        EP_AGORA,               EP_ATLAS,
EV_COG_AD3029LZ,        EV_COG_AD4050LZ,        FF1705_L151CC,
FF_LPC546XX,            FVP_MPS2_M0,            FVP_MPS2_M0P,
FVP_MPS2_M3,            FVP_MPS2_M4,            FVP_MPS2_M7,
GD32_F307VG,            GD32_F450ZI,            GR_LYCHEE,
GR_MANGO,               HEXIWEAR,               K22F,
K64F,                   K66F,                   K82F,
KL25Z,                  KL43Z,                  KL46Z,
KW41Z,                  LPC1114,                LPC1768,
LPC54114,               LPC546XX,               MAX32620FTHR,
MAX32625MBED,           MAX32625PICO,           MAX32630FTHR,
MIMXRT1050_EVK,         MOTE_L152RC,            MTS_DRAGONFLY_F411RE,
MTS_DRAGONFLY_F413RH,   MTS_DRAGONFLY_L471QG,   MTS_MDOT_F411RE,
NRF52840_DK,            NRF52_DK,               NUCLEO_F070RB,
NUCLEO_F072RB,          NUCLEO_F091RC,          NUCLEO_F103RB,
NUCLEO_F207ZG,          NUCLEO_F303K8,          NUCLEO_F303RE,
NUCLEO_F303ZE,          NUCLEO_F401RE,          NUCLEO_F411RE,
NUCLEO_F412ZG,          NUCLEO_F413ZH,          NUCLEO_F429ZI,
NUCLEO_F439ZI,          NUCLEO_F446RE,          NUCLEO_F446ZE,
NUCLEO_F746ZG,          NUCLEO_F756ZG,          NUCLEO_F767ZI,
NUCLEO_G031K8,          NUCLEO_G071RB,          NUCLEO_G431KB,
NUCLEO_G474RE,          NUCLEO_H743ZI2,         NUCLEO_H7A3ZI_Q,
NUCLEO_L073RZ,          NUCLEO_L152RE,          NUCLEO_L432KC,
NUCLEO_L433RC_P,        NUCLEO_L452RE_P,        NUCLEO_L476RG,
NUCLEO_L486RG,          NUCLEO_L496ZG,          NUCLEO_L496ZG_P,
NUCLEO_L4R5ZI,          NUCLEO_L4R5ZI_P,        NUCLEO_L552ZE_Q,
NUCLEO_WB55RG,          NUCLEO_WL55JC,          NUMAKER_IOT_M252,
NUMAKER_IOT_M263A,      NUMAKER_IOT_M487,       NUMAKER_PFM_M453,
NUMAKER_PFM_M487,       NUMAKER_PFM_NANO130,    NUMAKER_PFM_NUC472,
PORTENTA_H7_M4,         PORTENTA_H7_M7,         RHOMBIO_L476DMW1K,
RZ_A1H,                 S1SBP6A,                S5JS100,
SDP_K1,                 SDT32620B,              SDT32625B,
SDT52832B,              SDT64B,                 SFE_ARTEMIS,
SFE_ARTEMIS_ATP,        SFE_ARTEMIS_DK,         SFE_ARTEMIS_MODULE,
SFE_ARTEMIS_NANO,       SFE_ARTEMIS_THING_PLUS, SFE_EDGE,
SFE_EDGE2,              TB_SENSE_12,            TMPM46B,
TMPM4G9,                UHURU_RAVEN,            WIO_3G,
WIO_BG96,               WIO_EMW3166,            XDOT_L151CC
```
which looks like Murata Type ABZ is not supported in mbed-os.
The webpage description has no information about support versions and mbed availablity.
[image]

Can I run my program on this module?

Thanks for all replies and supports.

#### Target(s) affected by this defect ?
Murata Type ABZ

#### Toolchain(s) (name and version) displaying this defect ?
ARM Compiler 6.15

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio 1.4.1

#### How is this defect reproduced ? 
Always

adbridge, 2021-05-26:
@factoryal normally this is more a question for our forums https://forums.mbed.com/c/mbed-os/6 . There are a number of reasons why a target might be supported on Mbed 5 and not Mbed 6. It may no longer be supported by the partner or even just not capable of running Mbed 6. The Mbed team doesn't normally do the porting of targets, but we welcome contributions from the community. Please take a look here https://os.mbed.com/docs/mbed-os/v6.10/porting/index.html.

closed 2021-05-26 by adbridge

factoryal, 2021-05-26:
@adbridge Thanks for reply.
would you write some information at least it supports mbed OS 5 on webpage?
there's no information about that compared to other modules.
[image]

adbridge, 2021-05-26:
@MarceloSalazar who is responsible for these mbed enabled pages ?

MarceloSalazar, 2021-05-26:
Note the MURATA_ABZ is supported in Mbed OS 5.15 (see [here](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.15/targets/targets.json#L4288)) but not in Mbed OS 6.
I've added the corresponding 5.15 flag.

The modules are not Mbed Enabled. We'll update the pages shortly.

factoryal, 2021-05-26:
@MarceloSalazar Thanks. I tested on Mbed OS 5.15 today and it worked.
Is the partener different from DISCO-L072Z-LRWAN? It supports Mbed OS 6.
I tried this module because it is almost same as CMWX1ZZABZ-091 and there's no any comment on website as above. I didn't think they're that different.

MarceloSalazar, 2021-05-26:
@factoryal the MURATA_ABZ is based on STM32L082CZ and DISCO-L072Z-LRWAN is based on STM32L072CZ.
Different MCUs although part of the same L0 family. I don't know about CMWX1ZZABZ.

You may consider adding a custom board as described [here](https://os.mbed.com/docs/mbed-os/latest/porting/porting-custom-boards.html) and guidelines provided by ST [here](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/README.md#custom-boards).
It would be great if you share your custom configuration in the [Custom and community boards](https://forums.mbed.com/c/custom-and-community-boards/48) forum.

cc @ARMmbed/team-st-mcd

factoryal, 2021-05-26:
@MarceloSalazar 
MURATA_ABZ is CMWX1ZZABZ-078 and module used in DISCO-L072CZ-LRWAN is CMWX1ZZABZ-091.
Murata [says](https://www.murata.com/en-global/support/faqs/lpwa/lora/hardware/0003) there's no basic differences. 
It seems that it is runnable on MURATA_ABZ with selecting build target: DISCO-L072CZ-LRWAN1 .

factoryal, 2021-05-27:
the webpage has been updated. appreciate it!

hallard, 2021-07-16:
Interesting topic because each time I'm searching on differences between these Murata modules, it's a headache, so to keep trace somewhere, here are some precisions:
Murata ABZ [datasheet](https://wireless.murata.com/pub/RFM/data/type_abz.pdf) says that there have 2 modules:
[image]

difference between STM32L082 and STM32L072 and ST series states the only difference is that the L082 has HW AES encryption, not sure how it is (if it is) handled in mbed.
[image]

Please note there is now new version CMWX1ZZABZ-093 and  CMWX1ZZABZ-104 not mentioned in the datasheet on Murata web site so we can't find any information of L0 type it is. Love marketing guys...

And I can't help about the new Murata 1SJ module because even registered on murata support I can't have access to anything related to it without buying a module that have a registration code, amazing....

factoryal, 2021-07-16:
@hallard Hi.
This is interesting.
I can search CMWX1ZZABZ series on their website but I cannot find CMWX1ZZABZ-091 model. (discontinued?)
Is CMWX1ZZABZ-093 minor upgrade of CMWX1ZZABZ-091?
[image]

I think MCU used on these chip might be L072CZ or L082CZ because they have same physical size, flash size and memory size.

According to their website,...
- They confirmed CMWX1ZZABZ-078's MCU is STM32L082 and CMWX1ZZABZ-091 is STM32L072. (https://www.murata.com/en-global/support/faqs/lpwa/lora/hardware/0003)
[image]

- The CMWX1ZZABZ-104 has preloaded firmware for Semtech's cloud based device management.(it seems not for customizing firmware)
[image]

The only differences between STM32L082 and STM32L072 is AES hardware encryption feature and I can confirm that mbed program works on L082 developed on L072 settings. (<Murata Type ABZ> is not mbed enabled since mbed-os-6 but <DISCO-L072CZ-LRWAN1> supports)

I don't know why Murata website has so lacking information. that's really annoying.

https://wireless.murata.com/type-abz-078.html
https://wireless.murata.com/type-abz-091.html (404 ERROR)
https://wireless.murata.com/type-abz-093.html
https://wireless.murata.com/type-abz-104.html

thanks for comment :)

jeromecoutant, 2021-07-16:
> difference between STM32L082 and STM32L072 and ST series states the only difference is that the L082 has HW AES encryption, not sure how it is (if it is) handled in mbed.

Yes, difference between STM32L082 and STM32L072 is HW crypto.

You can enable it in Mbed-os with "MBEDTLS_CONFIG_HW_SUPPORT" macro

hallard, 2021-07-16:
@factoryal 
thanks for your answer, do you have any definition already done for mbed, may be we can add it to [stm32customtargets](https://github.com/ARMmbed/stm32customtargets) repo? I can do it if you push here your definitions files or I may just get back the old one removed from mbed os repos while going to v6.

Thanks

jeromecoutant, 2021-07-16:
Hi
Please have a look on https://github.com/ARMmbed/stm32customtargets/pull/11

factoryal, 2021-07-19:
@hallard
sorry. I'm just a beginner of mbed and don't know how to write my custom definitnions right now.
I just uploaded with target DISCO-L072CZ-LRWAN1.
I didn't know for the first time that there's two different models(-091, -078) because it worked so well without any different behaviors.

hallard, 2021-07-19:
@factoryal thanks, that is just what we wanted to know.

jeromecoutant, 2021-07-19:
As HW crypto is not enabled with STM32L082 within mbed, SW should be exactly the same for both L072 and L082.

An "easy" way to compare MCU capability could be to compare CMSIS file. Ex:
- https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L0/STM32Cube_FW/CMSIS/stm32l072xx.h and
- https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32L0/STM32Cube_FW/CMSIS/stm32l082xx.h