ARMmbed/mbed-os #14712: MAX32630FTHR throws Mutex error

idea--list, 2021-05-27:
### Description of defect

Have not been coding my MAX32630FTHR for some weeks now...
Today i tried to compile my code i regularly update onto this board, but i faced unexpected issue.
So i tried to compile the official blinky example with some minor additions this board needs to put 3V3 on the pins.

#### Target(s) affected by this defect ?
MAX3630FTHR

#### Toolchain(s) (name and version) displaying this defect ?
ARMC 6.15

#### What version of Mbed-os are you using (tag or sha) ?

OS V6.11 and V6.10

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio V1.4.1

#### How is this defect reproduced ? 

This code compiles&runs as expected up until OS V6.9 but throws mutex error on OS V6.10 and V6.11

```
#include "mbed.h"
#include "lib/max32630fthr/max32630fthr.h"

MAX32630FTHR max32630fthr(MAX32630FTHR::VIO_3V3);

// Blinking rate in milliseconds
#define BLINKING_RATE     500ms

int main()
{
    // Initialise the digital pin LED1 as an output
    DigitalOut led(LED1);

    while (true) {
        led = !led;
        ThisThread::sleep_for(BLINKING_RATE);
    }
} 
```

0xc0170, 2021-05-31:
What error is it (where is this coming from - what line is it invoking mutex error - locking in isr or is it something else)?

idea--list, 2021-06-08:
@0xc0170 This is all i get in terminal:

```
++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x0, Parameter error
Location: 0x7B15
Error Value: 0x0
Current Thread: main Id: 0x20001C70 Entry: 0x54D5 StackSize: 0x1000 StackMem: 0x20000650 SP: 0x2007FF44 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=MAX32630FTHR
-- MbedOS Error Info --
```

And that is what i get in Debug Console inside Mbed Studio:

```
0001021:INFO:board:Target type is max32630
0001057:INFO:dap:DP IDR = 0x2ba01477 (v1 rev2)
0001125:INFO:ap:AHB-AP#0 IDR = 0x24770011 (AHB-AP var1 rev2)
0001148:INFO:rom_table:AHB-AP#0 Class 0x1 ROM table #0 @ 0xe00ff000 (designer=43b part=4c4)
0001160:INFO:rom_table:[0]
0001166:INFO:rom_table:[1]
0001175:INFO:rom_table:[2]
0001178:INFO:cortex_m:CPU core #0 is Cortex-M4 r0p1
0001187:INFO:cortex_m:FPU present: FPv4-SP-D16-M
0001193:INFO:dwt:4 hardware watchpoints
0001199:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0001217:INFO:server:Semihost server started on port 4444 (core 0)
0001261:INFO:gdbserver:GDB server started on port 50000 (core 0)
Reading symbols from c:/Users/Peter/Mbed Programs/mbed-os-example-blinky/BUILD/MAX32630FTHR/ARMC6/mbed-os-example-blinky.elf...
warning: 
Loadable section "RW_IRAM1" outside of ELF segments
done.
0001422:INFO:gdbserver:Client connected to port 50000!
warning: Loadable section "RW_IRAM1" outside of ELF segments
```

idea--list, 2021-06-08:
Found #14465 introduced some changes to the Maxim boards. After reverting back /targets/TARGET_Maxim/TARGET_MAX32630/device/TOOLCHAIN_ARM_STD/MAX3263x.sct to the version that comes with OS V6.9 the blinky works again without mutex error. But for sure it would then affect bare-metal support which has been added with that merge.

I observed such changes several times and wonder why such things happen? I mean for sure the idea was to improve functionality and the new code may work much better in _theory_. But in practice even the most simple program fails to run with the newly merged code. So how new code is tested in general if not on the actual target?

ahmetalincak, 2022-03-31:
@0xc0170 #15251 fixes this.

closed 2022-03-31 by 0xc0170