ARMmbed/mbed-os #14714: Support stm32wl-lora-driver.rf-switch-ctl1..3 as NC

Niondir, 2021-05-28:
By Default the RF-Switch CTL1 to CTL3 lines are NC and will cause a HardFault. On my (custom) board, I only utilize a single Pin for the RF switch.
There could be a little documentation on the 3 Pins (or do I miss it somewhere) and they should be optional.

See comment in `void STM32WL_LoRaRadio::set_antenna_switch(RBI_Switch_TypeDef state)`:

```
// here we got to do ifdef for changing controls    
// as some pins might be NC
```

https://github.com/ARMmbed/mbed-os/blob/5f208ae90aadcb23aa307627814543ced4f99aa0/connectivity/drivers/lora/TARGET_STM32WL/STM32WL_LoRaRadio.cpp#L488

adbridge, 2021-05-28:
@Niondir thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

jeromecoutant, 2021-06-01:
Hi
I agree this function is not well coded.
I think I need to find a solution to move the **set_antenna_switch** function out of **STM32WL_LoRaRadio** class,
then, it could be implemented by each HW.

0xc0170, 2021-06-07:
@Niondir can you review  the fix  #14734? If it fixes it.

PR #14734 (2021-06-11): STM32WL LORA: HW specific out of STM32WL_LoRaRadio class
### Summary of changes 

Some code in STM32WL_LoRaRadio class was specific to NUCLEO_WL55JC HW,
and not general to STM32WL

It is then now compiled in another file.
Function is weak, and can be / has to be updated in each HW application

Should fix #14714

@Niondir  @ludoch-stm 

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2021-06-11 by 0xc0170