ARMmbed/mbed-os #14715: Feature: Add Cortex-A5 Support

Meano, 2021-05-30:
### Description of defect

Feature: Cortex-A5 Support
I am working on porting mbed-os on our Cortex-A5 SOC platform.
Could I create a PR to add Cortex-A5 support?

#### Target(s) affected by this defect ?
None, Testing on custom target.

#### Toolchain(s) (name and version) displaying this defect ?
I have tested changes by arm-none-eabi-10-2020Q4

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-tools

#### How is this defect reproduced ? 

The PR commit for mbed-os: https://github.com/Meano/mbed-os/commit/63ab58d4e5a0c4804bc99128c5a393f42583ca99
The PR commit for mbed-tools: https://github.com/Meano/mbed-tools/commit/d8d65a561d06c9977508747308b354db84d8735f

adbridge, 2021-05-30:
@Meano thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

Meano, 2021-05-30:
> @Meano thank you for raising this issue.Please take a look at the following comments:
>
> Could you add some more detail to the description? A good description should be at least 25 words.
>
> NOTE: If [...]

👌 

> @Meano thank you for raising this issue.Please take a look at the following comments:
>
> Could you add some more detail to the description? A good description should be at least 25 words.
>
> NOTE: If [...]

🆗 Updated

0xc0170, 2021-05-31:
It could be added so custom targets can inherit and use it.  Btw, what Cortex-A5 based MCU target you are using?

Meano, 2021-06-01:
> It could be added so custom targets can inherit and use it. Btw, what Cortex-A5 based MCU target you are using?

The chip is on the process of development. We port Mbed OS to customers who are more familiar with Cortex-M series.

adbridge, 2021-06-07:
@Meano yes please provide PRs to add support. The Mbed team don't actually port targets this is normally done by those requiring the board. I will close this issue accordingly. Any help/discussions can be made on any PRs that you raise. Thanks.

closed 2021-06-07 by adbridge