ARMmbed/mbed-os #14717: kv-store & block device in Mbed OS v6.x with TF-M

cyliangtw, 2021-05-31:
### Description of defect
For non TF-M platform, we could call standard Mbed API to operate kv-store & block device.
In Mbed OS v6.x with TF-M, such like as MUS-CA target, it could support secure storage but we can't access it by standard Mbed API. Actually, Mbed-TLS are migrated to support the partition of non-secure domain and TF-M part in OS v6.x. So, is there any plan on (1) how to store key or provision  (2) how to operate block device for storage ?

#### Target(s) affected by this defect ?
TF-M target, such like as MUS-CA

#### What version of Mbed-os are you using (tag or sha) ?

v6.x + TF-M

adbridge, 2021-06-03:
Closed/reopened to retrigger the github action for issue handling.

mbedmain, 2021-06-03:
@cyliangtw thank you for raising this issue.Please take a look at the following comments:

What toolchain(s) are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

cyliangtw, 2021-06-03:
Updated the issue to consider all the fields.

closed 2023-02-20 by cyliangtw