ARMmbed/mbed-os #14719: Cellular: recvfrom should return error if socket is closed

boraozgen, 2021-06-01:
### Description of defect

IMHO the socket receive function should return a error value (e.g. `NSAPI_ERROR_NO_SOCKET`) in case of a closed socket. Currently it returns 0, which corresponds to no error and no data:

https://github.com/ARMmbed/mbed-os/blob/0738aabb87476ae04ae9fae69f39250c2c9ce8ff/connectivity/cellular/source/framework/AT/AT_CellularStack.cpp#L333-L336

https://github.com/ARMmbed/mbed-os/blob/0738aabb87476ae04ae9fae69f39250c2c9ce8ff/connectivity/cellular/source/framework/AT/AT_CellularStack.cpp#L358-L361

Is there a particular reason for this?

@kjbracey-arm you probably would have an idea on this.

#### Target(s) affected by this defect ?

Cellular modems.

#### Toolchain(s) (name and version) displaying this defect ?

All.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.11.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

All.

#### How is this defect reproduced ? 

Design issue.

boraozgen, 2021-06-01:
Well as a counterargument [the POSIX article](https://pubs.opengroup.org/onlinepubs/9699919799/functions/recvfrom.html) indicates that 0 should be returned when the peer has closed the socket.

On the other hand there are other implementations (like [ARM Keil MDK](https://www.keil.com/pack/doc/mw6/Network/html/group__bsd__routines.html#gac6b00a29e8f1664d7ba4a70bfb92d9db)) which return specific error codes in this case.

kjbracey-arm, 2021-06-02:
*As a counterargument the POSIX article indicates that 0 should be returned when the peer has closed the socket.*

Nice try, but no, that's for indication of a successful peer close or shutdown on a stream (TCP) to indicate "end of stream".

IIRC, normally that would be returned exactly once, and further attempts to read would return errors.

If this is "recvfrom" in the Mbed OS socket API, then this would be for UDP (datagram) I believe, in which case "0" means "zero-length packet received".

Yes, this should be returning an error. Not totally sure what - maybe check other implementations. Depends what this "closed" condition means. If it was something that you could be in before "bind", then pre-bind, I think "would block" would be fine. If it was the result of the modem deciding to close, you'd want a real error code.

adbridge, 2021-06-03:
Closed/reopened to retrigger the github action for issue handling.