ARMmbed/mbed-os #14726: SX126X Select Pins

evandavey, 2021-06-02:
### Description of defect

The SX126X driver makes use of device, frequency and crystal select pins.  These are present in the EVB but not in any target using a SX126X module directly.  There should be an option to set the device, frequency or crystal configuration in macros, or the library split into separate modules for each device variant.

#### Target(s) affected by this defect ?

Any target wanting to use a SX126X module.

#### Toolchain(s) (name and version) displaying this defect ?

Any.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.5

#### How is this defect reproduced ? 

Use the SX126X library on a target other than the EVB board without dedicating additional pins for config selection.

0xc0170, 2021-06-02:
Can you make a pull request with configuration enabling these settings to be changed ?

How would you extend the mbed_lib.json file. The current version has this there:

```
{
    "name": "SX126X-lora-driver",
    "config": {
        "spi-frequency": {
        	"help": "SPI frequency, Default: 16 MHz",
        	"value": 16000000
        },
        "buffer-size": {
        	"help": "Max. buffer size the radio can handle, Default: 255 B",
        	"value": 255
        },
        "boost-rx": {
        	"help": "Increases sensitivity at the cost of power ~2mA for around ~3dB in sensitivity 0 = disabled, 1 = enabled",
        	"value": 0
        },
        "regulator-mode": {
        	"help": "Default: DCDC (low power, high BOM). Alternatively, LDO = 0. Check datasheet section 5.1 for more details",
        	"value": 1
        },
        "sleep-mode": {
        	"help": "Default: Cold start = 1, Warm start = 0. Check SleepMode.txt",
        	"value": 1
        },
        "standby-mode": {
        	"help": "Default: STDBY_RC = 0, STDBY_XOSC = 1",
        	"value": 0
        },
        "spi-mosi": {
            "value": "NC"
        },
        "spi-miso": {
            "value": "NC"
        },
        "spi-sclk": {
            "value": "NC"
        },
        "spi-cs": {
            "value": "NC"
        },
        "reset": {
            "value": "NC"
        },
        "dio1": {
            "value": "NC"
        },
        "busy": {
            "value": "NC"
        },
        "freq-select": {
            "value": "NC"
        },
        "device-select": {
            "value": "NC"
        },
        "crystal-select": {
            "value": "NC"
        },
        "ant-switch": {
            "value": "NC"
        }
    }
}

```

evandavey, 2021-06-02:
@0xc0170  - I have been using an implementation where I have:
```
       "device-variant": {
            "help": "Default: -1 = use device-select, SX1261, SX1262, SX1268",
            "value": -1
        },
        "freq-support": {
            "help": "Default: -1 = use freq-select, MATCHING_FREQ_169, MATCHING_FREQ_280, MATCHING_FREQ_434, MATCHING_FREQ_490, MATCHING_FREQ_780, MATCHING_FREQ_868, MATCHING_FREQ_915",
            "value": -1
        },
        "xtal": {
            "help": "Default: -1 = use crystal-select, TXCO = 0, XTAL = 1",
            "value": -1
        },
```

then use these in #if statements around get_frequency_support, get_device_variant and the __crystal_select.is_connected() && _crystal_select == 0 check but I will put these together as a pull request.  Since _freq_select and _dev_select are AnalogIn I had to wrap these as DigitalIn as well as leaving these NC was throwing errors.

adbridge, 2021-06-03:
Closed/reopened to retrigger the github action for issue handling.

PR #14730 (2021-07-21): Add config macros for device-variant, freq-support and xtal config

### Summary of changes 

Fixes #14726 by adding config macros for device-variant, freq-support and xtal config.
Settings these macros to -1, the selection pins are used.

#### Impact of changes 

#### Migration actions required
closed 2021-07-21 by 0xc0170