ARMmbed/mbed-os #14727: Peripheral Clock enabling before writing registers

chrJost, 2021-06-02:
### Description of defect

In some targets the registers for peripherals were written before the clock to the peripherals was activated. Therefore the changes to the registers were not done.

See https://github.com/ARMmbed/mbed-os/pull/14724 and https://github.com/ARMmbed/mbed-os/pull/14720

The remaining question is if this kind of defect can be detected automatically. It can be tedious to look through the start code of all devices by hand, in the case of the STM32L552 this was detected "by accident".

I would appreciate any input on this subject. Maybe some sort of guide on peripheral initialization?

#### Target(s) affected by this defect ?
STM32, maybe others as well

#### Toolchain(s) (name and version) displaying this defect ?
any

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.11.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
any

#### How is this defect reproduced ? 

Write a register and see if it actually changed something.

adbridge, 2021-06-03:
Closed/reopened to retrigger the github action for issue handling.

jeromecoutant, 2021-06-11:
ST_INTERNAL_REF 108872

jeromecoutant, 2021-10-05:
Hi
I think we could close this one as it is not related to mbed-os ?
Internal ST ticket is on going.
@0xc0170

closed 2021-10-05 by 0xc0170