ARMmbed/mbed-os #14731: mbed_die doesn't have access to LED1 when building with cmake

jrobeson, 2021-06-03:
### Description of defect
When compiling for my custom targets and NRF52840_DK I see:

```bash
 Building C object CMakeFiles/BLE_GattServer_CharacteristicUpdates.dir/mbed-os/platform/source/mbed_board.c.obj
../../../../mbed-os/platform/source/mbed_board.c: In function 'mbed_die':
../../../../mbed-os/platform/source/mbed_board.c:31:12: warning: unused variable 'led_err' [-Wunused-variable]
   31 |     gpio_t led_err;
```

and this is because `led_err` is only used if `LED1` is defined

#### Target(s) affected by this defect ?
NRF52840_DK but likely more

#### Toolchain(s) (name and version) displaying this defect ?
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

#### What version of Mbed-os are you using (tag or sha) ?
5fe4dafc349c4b0e93e3d2daeb933977c6851653

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-tools 7.15.0

#### How is this defect reproduced ? 
just build anything with NRF52 or perhaps other targets.

adbridge, 2021-06-03:
Closed/reopened to retrigger the github action for issue handling.

jrobeson, 2021-06-15:
Can somebody please verify whether this is a problem with other targets?  If it does happen with other targets, then `mbed_die` will never do the flashy flashy on crash for a lot of people.  If that's true, then the issue severity should be bumped.