ARMmbed/mbed-os #14758: STM32G474 InterruptIn callback not called

pilotak, 2021-06-10:
### Description of defect

When using 3 or more `InterruptIn` objects, one callback is never called - which one? that depends on the init order.
When using 2 objects everything is fine. I use this for detecting multiple button presses, please see the example below

```cpp
// callback on PB_2 never called
InterruptIn button[3] = {
    {PB_2, PullUp},
    {PC_5, PullUp},
    {PD_2, PullUp},
};

// callback on PB_2 never called
InterruptIn button[3] = {
    {PC_5, PullUp},
    {PB_2, PullUp},
    {PD_2, PullUp},
};

// callback on PB_2 never called
InterruptIn button[3] = {
    {PB_2, PullUp},
    {PD_2, PullUp},
    {PC_5, PullUp},
};

// callback on PD_2 never called
InterruptIn button[3] = {
    {PC_5, PullUp},
    {PD_2, PullUp},
    {PB_2, PullUp},
};

// callback on PD_2 never called
InterruptIn button[3] = {
    {PD_2, PullUp},
    {PB_2, PullUp},
    {PC_5, PullUp},
};

// callback on PD_2 never called
InterruptIn button[3] = {
    {PD_2, PullUp},
    {PC_5, PullUp},
    {PB_2, PullUp},
};
```

#### Target(s) affected by this defect ?
STM32G474

#### Toolchain(s) (name and version) displaying this defect ?
GCC-ARM 

#### What version of Mbed-os are you using (tag or sha) ?

aeaac0e70c76fbc5b753ee8c836e369edf1a7fb4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
Fo simplicity I used a `printf` in ISR which will give following error: `Error Message: Error - writing to a file in an ISR or critical section`. This way you can easilly see that the callback was called.

`mbed_app.json`
```json
{
  "target_overrides": {
    "*": {
      "mbed-trace.enable": true,
      "platform.error-all-threads-info": true,
      "platform.error-filename-capture-enabled": true,
      "platform.fatal-error-auto-reboot-enabled": true,
      "platform.stdio-convert-newlines": true,
      "platform.stdio-baud-rate": 115200,
      "platform.callback-nontrivial": true,
      "drivers.uart-serial-rxbuf-size": 2048,
      "drivers.uart-serial-txbuf-size": 256,
      "target.printf_lib": "std"
    },
    "NUCLEO_G474RE": {
      "target.stdio_uart_tx": "PB_3",
      "target.stdio_uart_rx": "PB_4",
      "target.clock_source": "USE_PLL_HSE_XTAL"
    }
  }
}
```

```cpp

#include "mbed.h"

InterruptIn button[3] = {
    {PC_5, PullUp},
    {PB_2, PullUp},
    {PD_2, PullUp},
};

void test() {
    printf("test\n");
}

int main() {
    printf("start\n");

    for (int i = 0; i < 3; i++) {
        button[i].fall(test);
    }

    while (1) {
        ThisThread::sleep_for(1s);
    }

    return 0;
}
```

LMESTM, 2021-06-10:
On STM32 external interrupts are share lines.
All PA_1, PB_1, PC_1 ... PG_1 share one single line 1
All PA_2, PB_2, PC_2 ... PG_2 share one single line 2
All PA_N, PB_N, PC_2 ... PG_N share one single line N

User should only consider using one EXTI per GPIO bank.
https://stm32f4-discovery.net/2014/08/stm32f4-external-interrupts-tutorial/
https://community.st.com/s/question/0D50X00009XkZsL/nvic-interrupts-on-two-ports-with-same-pin-number

@jeromecoutant 
I guess we may add an assert to raise an error at the object creation ?

pilotak, 2021-06-10:
Ok,  I didn't know that, so i had a luck on my previous boards.

If raising an error is possible it would be perfect! it looks like a bug otherwise

closed 2021-06-15 by pilotak

LMESTM, 2021-06-16:
@pilotak thanks we've added the request for raising an error in our to-do list ...