ARMmbed/mbed-os #14759: new mbedtls/timing.cpp creates double timing.o module

JojoS62, 2021-06-10:
### Description of defect

#14756 produces a linker warning because there exists already a timing.c:

`Object file timing.o is not unique! It could be made from: mbed-os/connectivity/mbedtls/source/timing.c mbed-os/connectivity/mbedtls/platform/src/timing.cpp`

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

b8781e527bc1931bf8f9481dfc63df6c4ce8229b

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli v1

#### How is this defect reproduced ? 

compile with mbed-cli v1

mbedmain, 2021-06-10:
@JojoS62 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-06-10:
Thanks for the report, what toolchain are you testing with?

Although this comes from mbed tools :( I see there method `detect_duplicates` in tools/resources/__init__.py. I recall there were problems in some IDE that might have a  problem with that, otherwise it should not be.

0xc0170, 2021-06-10:
Is this just a linker warning? From the code, it looks like it should be tool error (I wonder why CI did not catch this error).

JojoS62, 2021-06-10:
the binary image was created, and the target is running. But I'm not using the mbedtls, so the problem may have no effect.

PR #14760 (2021-06-11): mbedtls: Rename Mbed timing implementation

### Summary of changes 

We get a linker warning with the recently added timing module
implementation for Mbed. This is because there is Mbed TLS also ships a
file called timing.c, which we are including in Mbed OS also. With CLI
1, we get an error about unique object files because of the similarly
named implementation files.

    Object file timing.o is not unique! It could be made from: mbed-os/connectivity/mbedtls/source/timing.c mbed-os/connectivity/mbedtls/platform/src/timing.cpp

Rename the Mbed timing module implementation to timing_alt.cpp to [... cut]
commit 87d1992f (2021-06-11): mbedtls: Rename Mbed timing implementation

We get a linker warning with the recently added timing module
implementation for Mbed. This is because there is Mbed TLS also ships a
file called timing.c, which we are including in Mbed OS also. With CLI
1, we get an error about unique object files because of the similarly
named implementation files.

    Object file timing.o is not unique! It could be made from: mbed-os/connectivity/mbedtls/source/timing.c mbed-os/connectivity/mbedtls/platform/src/timing.cpp

Rename the Mbed timing module implementation to timing_mbed.cpp to avoid
this naming confl [... cut]
closed 2021-06-11 by 0xc0170