ARMmbed/mbed-os #14762: Musca B1/S1: memory layout headers out of date

LDong-Arm, 2021-06-10:
### Description of defect

The headers `flash_layout.h` and `region_defs.h` for TF-M targets are directly imported from the trusted-firmware-m repository and should be upgraded along with TF-M. But they are currently one version behind (v1.2 instead of v1.3) for [Musca B1](https://github.com/ARMmbed/mbed-os/tree/mbed-os-6.11.0/targets/TARGET_ARM_SSG/TARGET_MUSCA_B1/partition) and two versions behind (v1.1) for [Musca S1](https://github.com/ARMmbed/mbed-os/tree/mbed-os-6.11.0/targets/TARGET_ARM_SSG/TARGET_MUSCA_S1/partition) in Mbed OS. The TF-M project has already changed parts of the layout.

It happened, because both files are _not_ covered by the mbed-os-tf-m-regression-test's [import script](https://github.com/ARMmbed/mbed-os-tf-m-regression-tests/blob/master/tfm_ns_import.yaml). We left them out because of a workaround for the passing of the `BL2` macro to ARMC6, as explain in https://github.com/ARMmbed/mbed-os/commit/0ccf4db14310d13bbee7a72324db6203dd844974.

So we need to update those files to those from the latest release (TF-M v1.3 at the time of writing). It would be better to find a permanent solution to not need the `BL2` workaround, so we can add them to the import script.

#### Target(s) affected by this defect ?

ARM_MUSCA_S1
ARM_MUSCA_B1

#### Toolchain(s) (name and version) displaying this defect ?

N/A

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.11.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A

#### How is this defect reproduced ? 

Check the git histories of the files (linked above).

hazzlim, 2021-08-06:
Permanent solution (passing macros to armlink using `--predefine="<MACRO>"`) requires changes to frozen /tools.
This can be picked up following TF-M v1.4.0  upgrade, at which point BL2 macro workaround will need to be added to upgraded headers.

LDong-Arm, 2021-08-23:
Even though the BL2 workaround is still needed because of frozen tools, I think there's a better one.
Hardcoding it in `region_defs.h` prevents us from using our [automated script](https://github.com/ARMmbed/mbed-os-tf-m-regression-tests/blob/master/tfm_ns_import.yaml) to import this from from the trusted-firmware-m repo, which is why the headers became outdated. We should instead hardcode this macro in linker scripts that include this header, allowing the header itself to be unmodified and automatically importable.

Anyway, we'll make this change during the TF-M v1.4.0 upgrade.

closed 2021-09-15 by Patater