ARMmbed/mbed-os #14763: USBSerial doesn't work with connect_blocking = false on STM32F401

riaancillie, 2021-06-10:
### Description of defect

The USBSerial device on mbed-os-6.10.0 and older (also tested previous version) does not seem to initialize the USB peripheral when the `connect-blocking` parameter is set to `false` in the constructor. When parameter is set to 'true', everything works as expected: code is paused until the microcontroller is connected to a host. Setting the parameter to 'false' the expectation is that code will continue to run as normal and that the microcontroller can be connected at a later stage to host, but this is not the case with the STM32F401: setting the parameter to 'false' results in no USB device being enumerated from the host - as if no pull-up is even being applied to D+.

#### Target(s) affected by this defect ?
Tested with Nucleo F401RE (C-02 revision) 

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.10.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed Studio v1.4.1

#### How is this defect reproduced ? 
VBUS connected to PA9, D- and D+ connected directly from USB connector to PA11, PA12 respectively.

Create a blank program in Mbed Studio and update the mbed-os library to mbed-os-6.10.0

`mbed_app.json`
```
{
    "requires": ["bare-metal","drivers-usb","events"],
    "target_overrides": {
        "*": {
          "target.c_lib": "small"          
        },
        "NUCLEO_F401RE": {                    
          "target.device_has_add": ["USBDEVICE"]          
        }
      }
}
```

`main.cpp`
```
#include <mbed.h>
#include <UsbSerial.h>

USBSerial usb(false); 

int main() {
  while(1) {
    ThisThread::sleep_for(500ms);    
  }
}
```

#### Further information
Drilling into [USBSerial.h](https://github.com/ARMmbed/mbed-os/blob/master/drivers/usb/source/USBSerial.cpp) in the constructor implementation, adding `connect()` the `else` condition resolves the problem and allows the `connection_blocking = false` to work as expected (device connected to host at later time)

```
USBSerial::USBSerial(bool connect_blocking, uint16_t vendor_id, uint16_t product_id, uint16_t product_release):
    USBCDC(get_usb_phy(), vendor_id, product_id, product_release)
{
    _settings_changed_callback = 0;

    if (connect_blocking) {
        connect();
        wait_ready();
    } else {
        init();
        connect();  /// <---------------------------- I added this line and it seems to resolve the reported issue.
    }
}
```

jeromecoutant, 2021-06-11:
Hi
Just a quick check.... does #14736 change your status ?
Thx

riaancillie, 2021-06-11:
> Hi
> Just a quick check.... does #14736 change your status ?
> Thx

Hi @jeromecoutant 

I checked quickly. No, this did not resolve the issue. Also, please note that I switched to a different target sinced the last reported issue (F103 -> F401).

Methodology:
I cleaned the build folder and I added `__HAL_PCD_ENABLE(&hpcd);` as per #14736 and also some gargabe text. Compilation failed so I am sure that `USBPhy_STM32.cpp` is being compiled. I then removed the garbage text and tested again. USB device still not being detected. 

I then attached a debugger. `__HAL_PCD_ENABLE` in `USBPhyHw::init` is definitely being hit.

As an experiment I added `HAL_PCD_Start` directly below  `__HAL_PCD_ENABLE` and then the USB device is detected and working as expected. I verified with the debugger: Seems like `USBPhyHw::connect()` is never being called with `connect_blocking = false` and consequently `HAL_PCD_Start` is also not called.

MubeenHCLite, 2021-07-30:
PR https://github.com/ARMmbed/mbed-os/pull/14970

closed 2021-08-02 by 0xc0170