ARMmbed/mbed-os #14774: STM32F4 SPI 3-wire mode doesn't work correctly

vznncv, 2021-06-14:
### Description of defect

Method  `int SPI::write(const char *tx_buffer, int tx_length, char *rx_buffer, int rx_length)` doesn't work correctly with SPI 3-wire mode.

Actual behavior:

First byte of the `rx_buffer` after `int SPI::write(const char *tx_buffer, int tx_length, char *rx_buffer, int rx_length)` method invocation contains last byte of the previous transaction, when SPI is configured in the 3-wire mode. Actual bytes are shifted to next positions. The last byte are "moved" to the next read transaction.

Expected behavior:

`rx_buffer` contains bytes only from current transaction.

#### Target(s) affected by this defect ?

STM32F411CEU6 (custom board https://github.com/vznncv/TARGET_BLACKPILL_F411CE)

(probably all STM32F4 boards)

#### Toolchain(s) (name and version) displaying this defect ?

arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.11.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli v1.10.5

#### How is this defect reproduced ? 

Example of BMX160 sensor usage in 3-wire mode with STM32F411CEU6:

```
/**
 * SPI 3 Wire demo with BMX160 sensor.
 *
 * sensor: https://wiki.dfrobot.com/BMX160_9_Axis_Sensor_Module_SKU_SEN0373
 * board: https://github.com/vznncv/TARGET_BLACKPILL_F411CE/tree/v0.1.4
 * mbed-os version: 6.11.0
 */

#include "mbed.h"

#include "stm32f4xx_hal.h"

//
// Hardware pins
//
#define BMX160_SDX PB_15 // SPI MOSI pin (STM32 SPI2 instance)
#define BMX160_SCX PB_13 // SPI CLK pin (STM32 SPI2 instance)
#define BMX160_CSB PA_8 // SPI SSEL pin (STM32 SPI2 instance)

// enable/disable SPI 3 wire rx buffer cleanup
#define ENABLE_SPI_3WIRE_FIX 0

// helper macro to check return code
static char app_error_msg_buf[256];
#define CHECK_RET_CODE(expr) do {                                                                         \
    int err = expr;                                                                                       \
    if (err != 0) {                                                                                       \
        snprintf(app_error_msg_buf, 256, "expression \"%s\" has returned error code %i", #expr, err);     \
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_APPLICATION, MBED_ERROR_CODE_UNKNOWN), app_error_msg_buf); \
    }                                                                                                     \
} while(0);

//
// hardware interfaces
//
[... 291 lines cut ...]
| read from bmx160 register 0x71 | <- 0x00 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x71 | <- 0x00 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x71 | <- 0x00 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x6B | <- 0x01 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x6B | <- 0x01 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x6B | <- 0x01 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x00 | <- 0xD8 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x00 | <- 0xD8 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x00 | <- 0xD8 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x71 | <- 0x00 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x71 | <- 0x00 |
| 1. SPI RX buffer isn't empty. Clear it   |
| read from bmx160 register 0x71 | <- 0x00 |
====== Blink demo ======
```

0xc0170, 2021-06-15:
@ARMmbed/team-st-mcd Please review

LMESTM, 2021-06-28:
Hi, sorry for slow answer.

The 3 wires support has been added and tested with ST sensor below:
https://os.mbed.com/teams/ST-Expansion-SW-Team/code/X_NUCLEO_IKS01A2_SPI3W/
It was tested to work okay so we not sure what would have happened since then.

Discussing with the team who wrote this sensor driver, it seems you're using SPI3W to interface the Bosch Sensortec sensor which, maybe, differs from our ST sensors on SPI3W behavior. Maybe the SPI3W protocol can be slightly different from vendor to vendor. In our case, if sensor team remembers well, the component, before returning the data, sends back the register address. Could that make a difference ?

Also a quick look at the datasheet mentions: For single byte read as well as write operations, 16-bit protocols are used.
So in your test sequence, after the 1st write to 0x6B, shouldn't you read as well ?
| write to  bmx160 register 0x6B | -> 0x01 |

Not sure if that would solve the issue but my point is that I wonder if this may be due to protocol difference ?

vznncv, 2021-07-01:
Hi

> Maybe the SPI3W protocol can be slightly different from vendor to vendor.

3-Wire SPI must be same across all vendors like 4-Wire SPI, although high-level device protocols 
(consequence of read/write operations to accesses device registers) may be different even across devices 
of the same vendor. But SPI interface shouldn't care about high-level device protocols. It's task of the driver that
uses SPI interface.

> So in your test sequence, after the 1st write to 0x6B, shouldn't you read as well ?

No, 16-bit protocol means that you send byte with register address and then read or write register value. For write operations it includes sending of 2 bytes.

## extra issue investigation

I got an oscilloscope and captured SPI frames with [sigrock](https://sigrok.org/) to get extra clarification about this issue.

Test programs:

```
/**
 * SPI 3 Wire demo with BMX160 sensor.
 *
 * sensor: https://wiki.dfrobot.com/BMX160_9_Axis_Sensor_Module_SKU_SEN0373
 * board: https://github.com/vznncv/TARGET_BLACKPILL_F411CE/tree/v0.1.4
 * mbed-os version: 6.11.0
 */

#include "mbed.h"

#include "stm32f4xx_hal.h"

//
// Hardware pins
//
#define BMX160_SDX PB_15 // SPI MOSI pin (STM32 SPI2 instance)
#define BMX160_SCX PB_13 // SPI CLK pin (STM32 SPI2 instance)
#define BMX160_CSB PA_8 // SPI SSEL pin (STM32 SPI2 instance)

// helper macro to check return code
static char app_error_msg_buf[256];
#define CHECK_RET_CODE(expr) do {                                                                         \
    int err = expr;                                                                                       \
    if (err != 0) {                                                                                       \
        snprintf(app_error_msg_buf, 256, "expression \"%s\" has returned error code %i", #expr, err);     \
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_APPLICATION, MBED_ERROR_CODE_UNKNOWN), app_error_msg_buf); \
    }                                                                                                     \
} while(0);

//
// hardware interfaces
//
DigitalOut user_led(LED1, 1);
SPI bmx_spi(BMX160_SDX, NC, BMX160_SCX);
DigitalOut bmx_ssel(BMX160_CSB, NC);

// helper variables
int transaction_count = 0;

/**
 * Write to BMX160 register.
 *
 * @param reg register address
 * @param value register value
 * @return 0 on success, otherwise non-zero value
 */
int bmx_register_write(uint8_t reg, uint8_t value)
{
    bmx_ssel.write(0);

[... 133 lines cut ...]
I got the following results:

According documentation in "Bidirectional receive procedure" SPI starts generating clock signal continuously after SPI enabling
(see `20.3.5. Data transmission and reception procedures` in the reference manual), even if data isn't read from `DR` register. 
To stop bytes receiving you need to:

- Wait for the second to last occurrence of RXNE=1 (n–1)
- Then wait for one SPI clock cycle (using a software loop) before disabling the SPI (SPE=0)
- Then wait for the last RXNE=1 before entering the Halt mode (or disabling the peripheral clock)

 (see `20.3.8. Disabling the SPI` in the reference manual)

Whereas the code of `HAL_SPI_Receive` simply receives all n bytes and stop SPI. It causes that with some chance SPI may
start generating clock signal for n+1 byte.

So correct implementation of the `HAL_SPI_Receive` requires strict timings between (n-1) byte receiving and SPI disabling (especially with high SPI frequency). It implies the following limitations:
- all interrupts must be disabled during SPI data receiving to eliminate any delays
- the code gather data from SPI `DR` register should be short and fast to eliminate delays between (n-1) byte reading and SPI disabling. It may be difficult to create such code that will work with different optimization levels of compiler and different MCUs.
- additionally debugging may cause delays, incorrect driver behavior and strange bugs

### Summary

Current STM 3-Wire SPI hardware implementation doesn't provide reliable way to read bytes without dummy reads (or at least I don't find such approaches). So I think that it's better to drop support 3-Wire SPI from MbedOS as it has buggy implementation, or at least add some warning and cautions to documentation/code that STM 3-Wire SPI isn't able to read data reliable.

@LMESTM What do you think about it?

LMESTM, 2021-07-08:
@vznncv thanks a lot for this detailed analysis and explanation. 
What I think is that I will need to get feedback internally from experts to provide a definitive feedback.  
I'll try to come back to you asap !

LMESTM, 2021-07-08:
One more point: does that mean that your issue may be "solved" (worked-around) by requesting a slow  SPI frequency ?

LMESTM, 2021-07-08:
@vznncv  your analysis has been confirmed. The use case is actually document in the driver code here:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/STM32Cube_FW/STM32F4xx_HAL_Driver/stm32f4xx_hal_spi.c#L55

It means that if we want to work-around this issue in the mbed-os code, we need to add calls to 
   HAL_SPI_DeInit(handle);
   HAL_SPI_Init(handle);

probably here: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/stm_spi_api.c#L786
and here:  https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/stm_spi_api.c#L864

Would you mind giving it a try ?
I admit that it would probably degrade slightly performances but that might be worth it if this stabilize the transfers.

vznncv, 2021-07-10:
@LMESTM 

I have modified example to demonstrate burst read, as I noted that it's less stable: 
[bmx_3wire_error.cpp](https://github.com/ARMmbed/mbed-os/files/6795697/bmx_3wire_error.cpp.txt)

> One more point: does that mean that your issue may be "solved" (worked-around) by requesting a slow SPI frequency ?

Yes, it helps.

Demo uart output: [demo_200KHz_uart_output.txt](https://github.com/ARMmbed/mbed-os/files/6795701/demo_200KHz_uart_output.txt) 
Logical analyzer data: [demo_200KHz_sigrock.zip](https://github.com/ARMmbed/mbed-os/files/6795702/demo_200KHz_sigrock.zip)

But this solution has the following problems:

- I achieved stable communication only at 200 KHz. It's slower that I2C fast mode.
- If any interrupt occurs during communication (OS scheduler, ticker, etc.), I get invalid results.

> It means that if we want to work-around this issue in the mbed-os code, we need to add calls to
> HAL_SPI_DeInit(handle);
> HAL_SPI_Init(handle);
> probably here: [...]

I checked it. It doesn't help.

Demo uart output: [hal_reinit_uart_output.txt](https://github.com/ARMmbed/mbed-os/files/6795704/hal_reinit_uart_output.txt)  
Logical analyzer data: [hal_reinit_sigrok.zip](https://github.com/ARMmbed/mbed-os/files/6795706/hal_reinit_sigrok.zip)

### Extra investigation and workaround

After some experiments I found that timing between last byte receiving and SPI disabling is crucial. According
reference manual ("20.3.8 Disabling the SPI") it shouldn't be too small:

```
In master unidirectional receive-only mode (MSTR=1, BIDIMODE=0, RXONLY=1) or bidirectional receive mode (MSTR=1, BIDIMODE=1, BIDIOE=0)
This case must be managed in a particular way to ensure that the SPI does not initiate a new transfer. The sequence below is valid only for SPI Motorola configuration (FRF bit set to
0):
1. Wait for the second to last occurrence of RXNE=1 (n–1)
2. Then wait for one SPI clock cycle (using a software loop) before disabling the SPI (SPE=0)
3. Then wait for the last RXNE=1 before entering the Halt mode (or disabling the peripheral clock)
```

and it should be too high (otherwise next byte receiving is started).

So the following data receiving approach should work:

1. Configure SPI bidirectional receive mode.
2. Enter into critical section (disable all interrupts to eliminate any interrupt delays)
3. Enable SPI.
4. Wait one SPI clock cycle with software loop.
5. Disable SPI.
6. Exit from critical section.
7. Wait full byte receiving.
8. Read data for `DR` register.
9. If extra bytes should be read, then go to step 2.
10. All bytes are received.

The code that implements such approach (I have added it to `stm_spi_api.c`):
```
// extra includes
#include "mbed_critical.h"
#include "mbed_wait_api.h"

//
// other functions ...
//

/**
 * Receive SPI data in bidirectional receive mode.
 *
 * @param obj
[... 86 lines cut ...]
        }
    }

    return total;
}

//
// other functions ...
//
```

Such modification works correctly.

Demo uart output: [fix_1Mhz_uart_output.txt](https://github.com/ARMmbed/mbed-os/files/6795715/fix_1Mhz_uart_output.txt)  
Logical analyzer data: [fix_1MHz_sigrok.zip](https://github.com/ARMmbed/mbed-os/files/6795716/fix_1MHz_sigrok.zip)

It even works with higher SPI frequency (12.5 Mhz): [fix_12_5Mhz_uart_output.txt](https://github.com/ARMmbed/mbed-os/files/6795718/fix_12_5Mhz_uart_output.txt)  

(My oscilloscope cannot capture data at this frequency with *sigrok*, but according program output, data is received correctly)

`stm32f4_spi_master_one_wire_receive` function notes:
- I don't enable SPI for whole transaction (if multiple bytes should be received), because in this case all interrupts should be disabled during bytes receiving. This approach adds extra overhead, but prevents critical section retention during long time (several tens-hundreds microseconds in case of large transaction).
- `spi_master_write` doesn't require any modification as in 3-wire mode it implements only byte transmitting.

What do you think about such approach?

LMESTM, 2021-07-13:
Hi @vznncv 

Thanks for sharing a proposed solution. 
If we want to consider this kind of change, we should of course make it generic to work with all required families, not only F4.

Have you tested your proposed approach with a various number of read length ?
We could run a dedicated more detailed review in a specific Pull Request if you open one. 

Of course one major concern is how we will be able to test this proposal, as we don't have 3 WIRES SPI devices in our automated  regression test setup :-(

vznncv, 2021-07-15:
Hi @LMESTM

> Have you tested your proposed approach with a various number of read length ?

Yes I have tested it with 3 bytes reading (timestamp value from BMX160) and with 1 bytes reading (ordinary configuration registers).

> We could run a dedicated more detailed review in a specific Pull Request if you open one.

I'll prepare a pull request in 1-2 weeks (currently I'm busy). I'll try to implement generic STM family approach.

> Of course one major concern is how we will be able to test this proposal, as we don't have 3 WIRES SPI devices in our automated regression test setup :-(

How do you test ordinary (4 WIRES) SPI interface? (Do you use some sensor or something else?)

jeromecoutant, 2021-07-15:
> How do you test ordinary (4 WIRES) SPI interface? (Do you use some sensor or something else?)

In ST, we have 2 setup:
- https://github.com/ARMmbed/ci-test-shield: SD card
- https://github.com/ARMmbed/mbed-os/tree/master/hal/tests/TESTS/mbed_hal_fpga_ci_test_shield

LMESTM, 2021-07-15:
@vznncv thanks for your efforts, that really makes sense to allow everyone to benefit from the good work you've done.

LMESTM, 2021-07-15:
> @vznncv thanks for your efforts, that really makes sense to allow everyone to benefit from the good work you've done.
(and no problem, we can wait for a few weeks, sure :-) )

PR #14981 (2021-09-02): Fix STM32 SPI 3-wire (synchronous API)
### Summary of changes 

Fixes #14774

##### Problem

Almost all STM32 MCU families (except STM32H7) continuously generates clock signal, till SPI is disabled in 3-wire
receiving mode. It means that SPI should be disabled by software during last byte receiving.

It imposes the following limitations of the code that receive SPI data:

- all interrupts (even systick timer) should be disabled to prevent any accidental delays.
- the SPI receiving code should be "fast". Any extra error/state checks may cause that delay and SPI won't be disabled
  in time.

If  [... cut]
closed 2021-09-02 by 0xc0170