ARMmbed/mbed-os #14775: Support for LoRaWAN 1.1 on MCU_STM32WL

Niondir, 2021-06-15:
### Description of defect

We need to use LoRaWAN 1.1 in a commercial project. We have our own board based on `MCU_STM32WL`. Sadly the target is not supported in the branch.

I guess the solution is to just rebase the branch onto the current master (or my current revision).

There are at least some conflicts during rebasing. I would kindly ask the maintainer of the `feature-lorawan-1-1` Branch to perform a rebase to enable STM32WL support.

@rvolosatovs @kivaisan ?

#### Target(s) affected by this defect ?

```
"MCU_STM32WLE5xC" : {
		"inherits" : [
			"MCU_STM32WL"
		],
		"extra_labels_add": [
			"STM32WLE5xC"
		],
		"macros_add": [
			"STM32WLE5xx"
		],
		"mbed_rom_start": "0x8000000",
		"mbed_rom_size": "0x40000",
		"mbed_ram_start": "0x20000000",
		"mbed_ram_size": "0x10000",
        "bootloader_supported": true
	},
```

#### Toolchain(s) (name and version) displaying this defect ?

Any

#### What version of Mbed-os are you using (tag or sha) ?
https://github.com/ARMmbed/mbed-os/#c73413893fb98aaaeda74513c981ac68adc8645d

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

> $ mbed --version
> 1.10.5

#### How is this defect reproduced ? 

Compile the `feature-lorawan-1-1` with the target listed above.

kivaisan, 2021-06-15:
Hi @Niondir , sorry but I'm no longer working on mbed/lora. @pan- , @paul-szczepanek-arm can you help on this?

0xc0170, 2021-06-15:
@ARMmbed/mbed-os-connectivity can you rebase the branch?

I am looking at the changes on the feature branch. The conflicts might be in unittests (we updated unittests to modern CMake). The rest of changes are in lorawan code, not much changed in the last releases.

rvolosatovs, 2021-06-15:
Not working on LoRaWAN anymore either, sorry!

paul-szczepanek-arm, 2021-06-28:
Just a headsup that I'm working on rebasing the branch - most conflicts are caused by the unit testing overhaul. I still have to sort out the added tests.

jeromecoutant, 2021-06-30:
FYI @ludoch-stm

paul-szczepanek-arm, 2021-07-02:
I'm sorry to say this turned out to be a too long a job. The conflicts created are too complex to resolve without knowing the meaning of the code. I cannot do this without having this prioritised and scheduled by my superiors.

The original author of the code could possibly manually reapply the collapsed changes onto a new feature branch but that is not something I can do without first understanding the changes - which is not something I can do in my spare time.

(The aborted attempt https://github.com/paul-szczepanek-arm/mbed-os/pull/68)

Niondir, 2021-07-09:
@paul-szczepanek-arm Thanks for the effort and estimation.

We will needs this for a project with the "Deutsche Bahn" where we want to control the station clocks and sync time via LoRaWAN Broadcast. (see german page: https://www.deutschebahn.com/de/Digitalisierung/technologie/digitaler_alltag/lorawan-3232292)

If there is a way to prioritize this ticket it would help us very much.

0xc0170, 2021-07-09:
> We will needs this for a project with the "Deutsche Bahn" where we want to control the station clocks and sync time via LoRaWAN Broadcast. (see german page: https://www.deutschebahn.com/de/Digitalisierung/technologie/digitaler_alltag/lorawan-3232292

That sounds interesting! There are english pages (I did not know about their effort) https://www.deutschebahn.com/en/Digitalization/technology/innovations/lorawan-3520294 and https://www.deutschebahn.com/en/Digitalization/technology/New-Technology/internet_of_things-3520354

@Niondir You can take feature lorawan branch and apply there MCU_STM32WL - it should be way easier as the feature branch was postponed and it's now out of date unfortunately with master. Applying specific target code should unblock your work in the meanwhile.

niondir, 2021-07-20:
Yes, moving the MCU_STM32WL to the LoRa branch will probably be easier. I will have a look.

hallard, 2021-07-31:
Very interested by this one also
Remembering developed Deutsche Bah paxcounter hardware prototype with Klaus, should have some PCB somewhere in my office, that's funny

niondir, 2021-08-13:
A little update on my efforts.

I added STM32WL support to the LoRaWAN 1.1 branch and can compile my code now.

Next steps:
- Enable and Test LoRaWAN 1.1
- Test LoRaWAN Broadcasts

In case everything works fine, I would create a PR with my code.

I had to do some extra work, to make the STM32WL working on the lorawan-1-1 branch:

```
- Update mbedtls
- Update devices_key
- Update storage/kvstore
- Add platform/FEATURE_EXPERIMENTAL_API
- Add TARGET_STM32WL
```

All mentioned updates are not from the latest master, but from the revision my project is build on: d147abc3e556c58e5e343d34b729bc2192e18bd3

hallard, 2021-08-25:
@niondir, this is great stuff, any chance to test your changes or do we need to wait the PR?

niondir, 2021-09-04:
Sure there is a chance, but I actually need to move it to GitHub. I will let you know once it's available. I'm also looking forward for Feedback.

niondir, 2021-09-29:
Since a lot of features are missing, we will most likely switch to https://github.com/Lora-net/LoRaMac-node 

There are sample implementations for the STM32WL: https://github.com/TheThingsIndustries/generic-node-se/tree/develop/Software/app 

For using LoRaWAN 1.1 with mbed-os I would suggest to integrate the external LoRaMac-node repository using a thin wrapper (needs to be written).

niondir, 2021-09-29:
I provided my PR. I have to note that I was able to Join TTN but uplinks only worked in rare random cases. I did not invested much time in debugging since we are now planning to use the LoRaMac-node stack.

0xc0170, 2021-09-30:
As stated earlier, the branch was developed some time ago with different team who is no longer involved.

> For using LoRaWAN 1.1 with mbed-os I would suggest to integrate the external LoRaMac-node repository using a thin wrapper (needs to be written).

It might be better to do so. Thanks for the suggestion. 
The feature branch won't be updated (reasons provided above). I'll close this issue as "won't fix".

closed 2021-09-30 by 0xc0170

hallard, 2021-09-30:
integration of loramac node would be awesome, Is there any plan on that in mbed team ?

0xc0170, 2021-10-01:
> integration of loramac node would be awesome, Is there any plan on that in mbed team ?

No plans for this.

hallard, 2021-10-01:
So just to be sure, LoRaWAN support on mbed will stay "as his" with no update ? We moved to mbed because it's simple and easy, but if LoRaWAN support isn't done anymore it's not usefull for us because it's our main target.