ARMmbed/mbed-os #14784: Drop Python 2.7 support

Patater, 2021-06-16:
### Description of defect

Python 2.7 has been deprecated for quite some time now. We are looking to update some of the python libraries we ourselves depend on, and finding we can't because we are still supporting Python 2.7 but the libraries we depend on have already dropped support.

#### Target(s) affected by this defect ?

Python

#### Toolchain(s) (name and version) displaying this defect ?

Python 2.7

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99
86b85a3e84c7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed CLI 1
Mbed CLI 2

#### How is this defect reproduced ? 

Try to update prettytable or other python libraries. Watch pip fail due to dependency conflicts.

```
Failed to build cryptography
requests 2.20.1 has requirement urllib3<1.25,>=1.21.1, but you'll have urllib3 1.26.5 which is incompatible.
mbed-os-tools 1.8.6 has requirement PrettyTable<=1.0.1, but you'll have prettytable 2.1.0 which is incompatible.
mbed-ls 1.8.6 has requirement PrettyTable<=1.0.1, but you'll have prettytable 2.1.0 which is incompatible.
icetea 1.2.4 has requirement prettytable<1.0, but you'll have prettytable 2.1.0 which is incompatible.
```

Patater, 2021-06-16:
After we have Python 2.7 dropped, we can update prettytable and a few other dependencies, too.

Patater, 2021-08-12:
We can pin dependencies for python 2 to versions that are still compatible with python 2, allowing newer versions of dependencies used when python 3 is used. We can also drop testing with Python 2.

LDong-Arm, 2021-09-20:
It's also worth adding `cysecuretools==2.1.0` because a few Cypress targets need it in their post-build scripts but it doesn't support python2.7. (Note: The existing post-build script is not compatible with newer versions of this package so we fix it at 2.1.0)

For now, I'll raise a PR to add it to our Docker image, so we can test the Cypress TF-M target in the PSA example.