ARMmbed/mbed-os #14791: Move events stubs to events/tests/UNITTESTS

rajkan01, 2021-06-17:
### Description of defect

To move events stubs to `events/tests/UNITTESTS` directory so the stubs live with the associated component

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

Latest

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
N/A

mbedmain, 2021-06-17:
@rajkan01 thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

LDong-Arm, 2021-07-01:
Reopening this issue - the PR isn't merged yet

PR #14826 (2021-06-25): CMake: unit-tests: Make events UNITTESTS only depend on headers it uses

### Summary of changes 
fixes #14791 
Previously the events unit tests depended on `mbed-headers`, which
is a collection of all available headers in mbed-os and these changes make drivers unit test only depend on the headers it's using.

#### Impact of changes 
None.

#### Migration actions required
PR #14823 (2021-07-12): Move events stubs to events dir

### Summary of changes 
fixes #14791 

- This PR makes the mbed-stubs-events library more self-contained by removing the dependencies on the monolithic header libraries we created to make implementing the stubs easier. We also move the files to the events directory so the stubs live with the associated component.
- Add mbed-headers-events library inside CMake of doubles
- Move events stubs headers from mbed-stubs-headers library to mbed-stubs-events library.
- Updated connectivity unit test CMake to use mbed-stubs-events-headers as it relies on events stub head [... cut]
closed 2021-07-12 by 0xc0170