ARMmbed/mbed-os #14809: OS_THREAD_LIBSPACE_NUM warning when application is defining thread-num

jeromecoutant, 2021-06-21:
### Description of defect

Issue seen while compiling mbed-os-tf-m-regression-tests

This application is defining: "rtos.thread-num": 9,
https://github.com/ARMmbed/mbed-os-tf-m-regression-tests/blob/master/mbed_app.json#L13

==> mbed_config.h:#define MBED_CONF_RTOS_THREAD_NUM 9 // set by application[*]

In that case, OS_THREAD_LIBSPACE_NUM is defined2 times:

https://github.com/ARMmbed/mbed-os/blob/master/cmsis/device/rtos/include/mbed_rtx_conf.h
#define OS_THREAD_NUM     MBED_CONF_RTOS_THREAD_NUM
#define OS_THREAD_LIBSPACE_NUM      4

https://github.com/ARMmbed/mbed-os/blob/master/cmsis/device/rtos/include/mbed_rtx_conf.h
#define OS_THREAD_LIBSPACE_NUM      OS_THREAD_NUM
==> 9 in our case, different than 4...

@Patater 

#### Target(s) affected by this defect ?

ALL

#### Toolchain(s) (name and version) displaying this defect ?

ALL

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.11.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ? 

See description

0xc0170, 2021-06-22:
I do not follow. What is the warning, where does it come from?

mbed_rtx_conf.h - this checks if MBED_CONF_RTOS_THREAD_NUM is not zero and applies its value to RTX macro value (LIBSPACE_NUM).

jeromecoutant, 2021-06-28:
> I do not follow. What is the warning, where does it come from?

In your mbed_app.json file, add:
````
    "target_overrides": {
        "*": {
            "rtos.thread-num": 9,
````

You will get the error:
````
Compile [  0.3%]: rtx_evr.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  0.4%]: rtx_lib.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  0.5%]: irq_cm4f.S
Compile [  0.6%]: rtx_delay.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  0.7%]: rtx_memory.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  0.9%]: rtx_evflags.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  1.0%]: rtx_kernel.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  1.1%]: rtx_mutex.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  1.2%]: rtx_mempool.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  1.3%]: rtx_system.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  1.4%]: rtx_msgqueue.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
Compile [  1.5%]: rtx_semaphore.c
[Warning] RTX_Config.h@575,9: 'OS_THREAD_LIBSPACE_NUM' macro redefined [-Wmacro-redefined]
````

0xc0170, 2021-06-28:
Thanks, this is clear now, we will need to review.