ARMmbed/mbed-os #14810: Enabling LSE with USE_PLL_HSE_XTAL clock source on an STM32L475xG-based target

g-arjones, 2021-06-21:
### Description of defect

I've written a custom target for a board with an STM32L475RGT. There's nothing fancy, no external peripherals but we do have an LSE and HSE. Looking into how those get initialized in the base target, [I noticed that apparently LSE is only enabled if `CLOCK_SOURCE` is `USE_PLL_MSI`](https://github.com/ARMmbed/mbed-os/blob/cecc47b4a53951527dd3f670465c8566396ad101/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L475xG/system_clock.c#L245). In both `SetSysClock_PLL_HSI()` and `SetSysClock_PLL_HSE()`, `RCC_OscInitTypeDef.LSEState` is left uninitialized. Is this intentional?

#### Target(s) affected by this defect ?

All targets inheriting from `MCU_STM32L475xG` and possibly more from the STM32L4 family.

#### Toolchain(s) (name and version) displaying this defect ?

ALL

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.12.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 2

#### How is this defect reproduced ? 

Static code analysis

0xc0170, 2021-06-22:
@ARMmbed/team-st-mcd please review

jeromecoutant, 2021-06-22:
Hi
LSE is used for MSI auto calibration.
So if you are using HSI or HSE for system clock, you don't need LSE there. (LSE will be enabled during LPTICKER or RTC)

closed 2021-06-22 by g-arjones