ARMmbed/mbed-os #14811: RTX5 event recorder events

boraozgen, 2021-06-22:
### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

mbedmain, 2021-06-22:
@boraozgen thank you for raising this issue.Please take a look at the following comments:

Could you add some more detail to the description? A good description should be at least 25 words.
What target(s) are you using?
What toolchain(s) are you using?
What Mbed OS version are you using?
It would help if you could also specify the versions of any tools you are using?
 How can we reproduce your issue?

NOTE: If there are fields which are not applicable then please just add 'n/a' or 'None'. This indicates to us that at least all the fields have been considered.
Please update the issue header with the missing information.

0xc0170, 2021-06-22:
Usually you check if a macro is defined then it's value so it can be enabled/disabled with a value. Anyone using it would check if it's defined, there is something somewhere already defining its value.
My initial suspicion is their event assumption is it is up to an application to choose this (final application, not something like Mbed OS). 

I would go upstream to ask about the use if it's just !defined(MACRO_DISABLE) how this can be configured.

boraozgen, 2021-06-23:
A look at rtx_evr.h shows that DISABLE macros are our only option, as disabling groups like OS_EVR_THREAD would also disable the error hooks which Mbed uses internally.

How about a single mbed_lib.json flag like `rtos.enable-all-events` which turns off all the disables?

0xc0170, 2021-06-23:
> How about a single mbed_lib.json flag like rtos.enable-all-events which turns off all the disables? As trace analyzers are the only use case I can think of, enabling all of the flags would be OK for debugging purposes.

That makes sense. It would comment out all we have disabled and a user can then select what to disable. 

I would add rtx to the name. rtos.enable-all-rtx-events. we can add to the help text - all rtx events that Mbed OS disables are enabled and a user can select what to disable instead.

Can you send a PR?

closed 2021-07-20 by 0xc0170