ARMmbed/mbed-os #14841: Add cmake-format check to CI

rwalton-arm, 2021-06-29:
### Description of defect

Currently the formatting style we follow for CMake files is ambiguous. We have various indentation rules we seem to require but they are only enforced in manual code review, and suggestions given aren't always consistent.

We can use the [cmake-format](https://pypi.org/project/cmake-format/) tool to automatically format CMake files. We should add a check to the CI, using this tool, to enforce a consistent formatting style on PRs that touch CMake files.

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ?

0xc0170, 2021-06-29:
There was a proposal that could be a good base: https://github.com/ARMmbed/mbed-os/pull/14011

cc @ladislas

ladislas, 2021-06-29:
haven't had time yet to test the tool, sorry about that :(

rwalton-arm, 2021-06-30:
If we use [pre-commit](https://pre-commit.com/) to manage the tool, enabling the check would be as simple as following the [cmake-format-precommit example](https://github.com/cheshirekow/cmake-format-precommit). Then in travis we would add `apt install pre-commit && pre-commit run --all-files`. The tricky bit will be agreeing on the formatting style we want to use (`cmake-format` has many [configurable settings](https://cmake-format.readthedocs.io/en/latest/configuration.html)). An added benefit of using `pre-commit` is it becomes very easy to add new PR checks, and there are many pre-commit compatible "hooks" available which could be useful for mbed-os.