ARMmbed/mbed-os #14848: EventQueue enqeues wrong this pointer when compiler optimization -Os is enabled

Niondir, 2021-07-01:
### Description of defect

Compiler optimization "-Os" in "--profile release" leads to wrong this point in `EvenQueue` event call.
 
Using the LoRaWAN stack in function `void LoRaWANStack::process_reception_timeout(bool is_timeout)` the this pointer is broken.

Further analysis shows that this is only the case when compiling with the release profile. The error happens already in `int equeue_post(equeue_t *q, void (*cb)(void *), void *p)` where the event data (holding the this pointer) is already broken.
Called by:

```
void LoRaWANStack::rx_timeout_interrupt_handler(void)
{
    const int ret = _queue->call(this, &LoRaWANStack::process_reception_timeout,
                                 true);
    MBED_ASSERT(ret != 0);
    (void)ret;
}
```

#### Target(s) affected by this defect ?

Custom Target:

```
{
	"MCU_STM32WLE5xC" : {
		"inherits" : [
			"MCU_STM32WL"
		],
		"extra_labels_add": [
			"STM32WLE5xC"
		],
		"macros_add": [
			"STM32WLE5xx"
		],
		"mbed_rom_start": "0x8000000",
		"mbed_rom_size": "0x40000",
		"mbed_ram_start": "0x20000000",
		"mbed_ram_size": "0x10000",
        "bootloader_supported": true,
		"sectors": [
			[
				134217728,
				2048
			]
		],
		"config": {
			"lse_bypass": {
				"help": "1 to use an oscillator (not a crystal) on 32k LSE",
				"value": "1"
			}
		}
	},
	"CLOCKMODULE" : {
		"inherits": [
			"MCU_STM32WLE5xC"
		],
		"device_name": "STM32WLE5JCIx",
		"OUTPUT_EXT": "hex"
	}
}
```

#### Toolchain(s) (name and version) displaying this defect ?

* arm-none-eabi-g++.exe (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)

#### What version of Mbed-os are you using (tag or sha) ?

* mbed-os-6.9.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

$ mbed --version
1.10.5

Building with CMake after `mbed export -i cmake_gcc_arm --profile release`

#### How is this defect reproduced ? 

See description above. I can't give the full sourcecode to reproduce right now. But I'm happy to test fixes.

I also need to update arm mbed and test again - but that needs further changes in my project and will follow later.

pan-, 2021-07-01:
Two questions: 

- Is the `this` pointer in `rx_timeout_interrupt_handler` valid ?
- Is it the `this` calling `process_reception_timeout` always invalid or does it become invalid after some operation ?

Niondir, 2021-07-03:
@pan- 

In `rx_timeout_interrupt_handler` the pointer was correct, I was following the calls through the call stack down to the enqueue. Somewhere on the way down the debugger states that `this` was optimized out. In the end a wrong address was written to the queue.

I disabled all other threads, I'm pretty sure it's not overwritten "in between" due to some concurrent code or stack overflow.